bluekitchen/btstack #397: If peer l2cap connect request received before read encrption key size response, stack will be re-authenticate again and link will be failed

nick1zhao, 2021-12-24:
**Describe the bug**
If peer l2cap connect request received before read encrption key size response, stack will be re-authenticate again and link will be failed

**To Reproduce**
Steps to reproduce the behavior:
1. Run example '...'
2. Connect to/from remove device '...'
3. Start action '...'

**Expected behavior**
A clear and concise description of what you expected to happen.

**HCI Packet Logs**
For all Bluetooth issues, please provide a full packet log that includes initial pairing. You need to zip the packet log as GitHub does not know how to handle .pklg files. 

How to get packet log depends on the port:
- For desktop ports, full packet logging is enabled by default and the packet log is available as `/tmp/hci_dump.pklg` (Mac/Linux), or `hci_dump.pklg' (Windows).
- For embedded ports, you can enable #define LOG_INFO in btstack_config.h and uncomment the call to 'hci_dump_init(..)' in the main() function to enable logging over UART Console or Segger RTT. Please capture text output and process using `btstack/tool/create_packet_log.py`. The tool will generate the text file into a .pklg file which can be analysed with Wireshark or Apple's PacketLogger tool.
- For ESP32: BTstack's `port/esp32/integrate.py` script copies the current version of BTstack into the esp-idf folder. To enable packet logging, you need to edit esp-idf/components/btstack/main.c

**Environment:  (please complete the following information):**
 - Current BTstack branch: [e.g. develop]
 - Bluetooth Controller [e.g. CSR 8510, TI CC2564C, ...]
 - Remote device: [e.g. iPhone X with iOS 13.3]

**Additional context**
Add any other context about the problem here.

mringwal, 2021-12-24:
Hi. BTstack triggers authentication after encryption complete for BR/EDR Legacy Connections (non SC) to reduce/avoid risk from BIAS attacks - that's the expected behaviour. The link shouldn't fail, though. Does this happens every time, or is it some kind or race condition?

What Bluetooth Controller do you use? What Bluetooth Controller/Stack is running on the remote side?
Could you provide HCI logs? Incoming connections work for me. Without logs and without being able to reproduce, there's no much we can do.

nick1zhao, 2021-12-25:
Hi,

HCI log as attached.  Test device is iPhone 7 (iOS 14.2)

Controller is ourself which based on RW 5.2 IP

mringwal, 2021-12-25:
Hi Nick. I don't see the log attached here, could you check on GitHub Issue tracker that it's there (looks like attachments won"t work for email replies)?
You may send it to me as well matthias@bluekitchen-gmbh.com

mringwal, 2022-01-17:
Hi Nick. Have you been able to resolve this somehow? I was not able to open your logs with Front Line's "View CFA" tool..

mringwal, 2022-08-20:
Timeout

closed 2022-08-20 by mringwal