bluekitchen/btstack #401: Incoming connections not available while gap_inquiry is on

ricardoquesada, 2022-01-20:
**Describe the bug**

Incoming connections cannot be detected while `gap_inquiry_start()` is running.

Not sure whether this is "working as intended", but seems to be a bug.
Or at least there should be a flag that toggles incoming connections while `gap_inquiry_start()` is on.

The use case is the following:

- have a HID host
- Host should auto-connect to certain devices (in my particular case: gamepads)
- But it must also allows incoming connections from gamepads: it could be a gamepad reconnect, or in some cases like the DualShock3 gamepad, it cannot be discovered. Only works when the gamepad starts the connection.

The work around that I found is to do:
 - a) call `gap_inquiry_start(1); // minum invertal possible` to discover to gamepads
 - b) do a pause of 1 second to allow incoming connections
 - c) goto a)

**To Reproduce**

The easiest way to reproduce it is by running the `hid_host_demo` with the following change:

- Call `gap_inquiry_start(30);` somewhere in the code. I did it line 272, inside the `BTSTACK_EVENT_STATE` event.
  -  here: https://github.com/bluekitchen/btstack/blob/d778e7488c62d361fc176e8ae40c0d6bc8b00089/example/hid_host_demo.c#L274
  -  and remove the `hid_host_connect()` code.
 
I was able to reproduce it with a DualShock3 gamepad... but any device that initiates a HID connect should be able to trigger.
You will notice that when:
  - `gap_inquiry_start()` is running, the gamepad cannot connect.
  - once the `gap_inquiry_start()` finishes, the connection can be established.

**Expected behavior**

btstack should be able to receive incoming connections while `gap_inquiry_start()` is running.

**HCI Packet Logs**

**Environment:  (please complete the following information):**
 - Current BTstack branch: develop and master branch
 - Bluetooth Controller: tested both with libusb (linux) and in a ESP32

[with_and_without_gap_inquiry.zip](https://github.com/bluekitchen/btstack/files/7907183/with_and_without_gap_inquiry.zip)

ricardoquesada, 2022-01-20:
I read a bit some HCI docs, and it is not clear to me whether this is a limitation of the HCI spec or not... in any case, it would be great to support it in case it is feasible.

mringwal, 2022-01-21:
@ricardoquesada It's not a limitation of the HCI spec as far as I know - I would have expected this to work. Could you try the same situation with a different Bluetooth Controller, e.g. USB Bluetooth dongle?

In general, the Controller should be able to do both Page Scan (for incoming connections) and Inquiry (to find devices) at the same time, at least I don't see a fundamental issue with that.

For incoming connections, Page Scan is relevant. Could you play with the gap_set_page_scan_activity(..) to increase the time the Controller is in Page Scan mode? There's also a gap_set_page_scan_type() which can be change from default to interlaced. I don't know the lower details of these yet, they are hidden in the Core Spec somewhere.

If tweaking these doesn't work but it works on a different controller (e.g. the USB dongle), please file an issue with the esp-idf project including both logs and link it here.

ricardoquesada, 2022-01-21:
Many thanks for the info.

yes, I tested it also with the CSR8510 A10 dongle with same results.
Should I try another dongle?

I looked at the spec:
  - 7.3.18: Write Scan enable command: I'm `gap_connectable_control(1);` `gap_discoverable_control(1);` before starting the scan
  - 7.3.20: Write Page Activity command: I didn't fully understand how it works, but tried with different values, 
  - 7.3.52: Write Page Scan Type command: tried with both options following the spec... although I don't fully understand what I was doing.... 

...and I didn't notice any change. Incoming connections doesn't occur while the "scan" is on :(

Any other idea? 
Many thanks!

BTW: Which dongle do you recommend? One that you know is fully compliant with the BR/ERD 4.2 spec (or 5.0).
Thanks!

ricardoquesada, 2022-01-22:
mmm... I don't think it is related to btstack.

I repeated the same scenario on Linux using bluez stack using the KDE "Bluetooth settings" application.

 - reset HCI by Disabling / Enabling Bluetooth
 - start inquiry by pressing "+" on the "Add new bluetooth devices"
 - immediately try to connect the gamepad. It is a DualShock3. It does not responds to inquiries. It will try to automatically connect to the Address that it is stored in its flash.
 - ... nothing happens...
 - ~10 seconds later the "inquiry complete" event is received
 - ...and right after the "inquiry complete" event the DualShock3 the "Connect Request" event is received.

So it seems that while the "Inquiry" is in progress no "Connect Request" are received (bluez and btstack behave the same way).
Not sure whether this can be "fixed"  with the "set event filter" command (chapter 7.3.3 of the spec). It has an "auto accept flag".
Although I noticed that it is not implemented in btstack.

Any chance that you can do a similar experiment with your dongle?... I'd like to know weather this is a Bluetooth "feature"... or just a bug in my dongle (and in ESP32).

Thanks!

PS: Attaching the Linux capture with bluez in case you are curious.
[linux-capture.zip](https://github.com/bluekitchen/btstack/files/7917803/linux-capture.zip)

mringwal, 2022-01-22:
@ricardoquesada Of course! From the logs it's clear that BTstack doesn't to anything wrong (as there's no Connection Request reported). My favorite USB dongle also is the CSR8510 (got a dozen of them from AliExpress at one point). I have others as well.

Around 2007, when Bluetooth v1.1 was around, we had a 80 nodes Bluetooth network and I think the main 'net forming" algorithm was "discover for x time and if you find another node, connect to it". Just reading up, later we switched to a form of tree building.

I don't know the details of the inquiry and the page scan algorithms but would assume that inquiry should not use 100% of air time, allowing the page scan mechanism to answer incoming connection requests. Interestingly, while there are commands to configure Page Scan and Inquiry Scan, there isn't one for Inquiry itself (and the start inquiry only has the LAP as parameter).

Hm.. the next best idea would be to periodically do an inquiry for a short amount of time - as you've already did.
On the TI CC2564, there's a vendor command that can be used to prioritize existing connections to avoid audio drop outs. Maybe that would also allow for incoming connection requests.

As you see, I don't have a clear answer either, sorry. Inquiry is a heavy procedure, so it generally is only started under user control, where we don't expect incoming connections per se. Could you integrate that into your design - some kind of "trigger connection to new gamepads"?

mringwal, 2022-01-22:
btw... if you want to auto-connect to known devices, you directly try to connect instead of doing an inquiry to see if the device is around.

ricardoquesada, 2022-01-22:
Thanks!
I have the same CSR8510 dongle.
I'll keep playing a bit with the different scan / page / inquiry commands.

Also, I'll try to play a bit with "set_event_filter" + auto connect.
Would you accept a patch for that? Anything in particular that I should do  when preparing the patch?

And yes, as a last resort, I thought of adding a "enter into reconnect mode" that a user can activate, where I stop the running inquiry.

Unfortunately, in my scenario, the host should not start the "connect to known devices"... that would be confusing for the users.

Many thanks!

mringwal, 2022-01-24:
hi. Feel free to try the auto connect feature, but I don't see a reason why this would make a difference. The controller seems to not receive (as in, it's not listening to) the Connection Request.

Any PRs should always be against the current develop branch in general. I'm not interested in a patch for auto-connect, as the stack needs to track incoming connections anyway, there's an incoming connection filter function, and we should not accept connection if we don't have free hci connection structs.

What about a physical button on your device that triggers the Inquiry + connect to found devices, while inquiry is off otherwise? This would be easy to explain to the user: "to pair new device press button x on our device and enter pairing mode on the other device"

ricardoquesada, 2022-01-24:
Thanks for the help Matthias!

I was referring to send you a patch for the "set event filter" HCI command. (Bluetooth Spec 4.2 , Vol 2, Part E, Section 7.3.3, or just search for "HCI_Set_Event_filter" in the PDF).

This command allows the controller to filter certain connections or inquiries based on Class of Device or BD_ADDR.
And it has an "auto accept" flag.

ricardoquesada, 2022-01-31:
Aha... so the "scan + pause" that I was doing manually can be achieved with the "periodic scan mode" command. It seems that it does the same (or similar) thing.
Sent a Pull-request for such command: https://github.com/bluekitchen/btstack/pull/402

ricardoquesada, 2022-02-01:
closed since it is not related to BTstack

closed 2022-02-01 by ricardoquesada