bluekitchen/btstack #403: esp32: fails to compile when using -O2 (strcat)

solhuebner, 2022-02-03:
**Describe the bug**
btstack fails to compile due to a warning that is treated as an error when "Compiler Options" -> "Optimization Level" -> "Optimize for performance (-O2)" is chosen.

**To Reproduce**
1. Use ESP-IDF v4.4
2. idf.py menuconfig
3. Select "Compiler Options" -> "Optimization Level" -> "Optimize for performance (-O2)"
4. Compile: idf.py build

```
In function 'pbap_client_process_vcard_list_body',
    inlined from 'pbap_client_parser_callback_get_operation$part$12' at ../components/btstack/src/classic/pbap_client.c:548:21,
    inlined from 'pbap_client_parser_callback_get_operation' at ../components/btstack/src/classic/pbap_client.c:528:13:
../components/btstack/src/classic/pbap_client.c:485:21: error: 'strcat' accessing between 162 and 2147483497 bytes at offsets 309 and 148 may overlap 1 byte at offset 309 [-Werror=restrict]       
                     strcat(pbap_client->parser_handle, pbap_client->xml_parser.data);
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../components/btstack/src/classic/pbap_client.c:476:21: error: 'strcat' accessing between 130 and 2147483497 bytes at offsets 277 and 148 may overlap 1 byte at offset 277 [-Werror=restrict]       
                     strcat(pbap_client->parser_name, pbap_client->xml_parser.data);
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

**Expected behavior**
It should compile fine ;)

**Environment:  (please complete the following information):**
 - Current BTstack branch: develop
 - Bluetooth Controller ESP32
 - Remote device: Windows 10

**Additional context**
A similar issue like #393 but another function :(

mringwal, 2022-02-03:
Lol.. these 'old' string functions will keep haunting us... at least I can only see these two calls in the core library. Will have a look.

mringwal, 2022-02-15:
Fixed on develop.

closed 2022-02-15 by mringwal