bluekitchen/btstack #404: CSR8311 with stm32f4-discovery can not change baud

xihua13104, 2022-02-10:
I use csr8311 with stm32f4-discovery to run this BT stack, i want to change BT uart buad to 921600 after send csr warm boot.
But it doesn't work, the next hci_reset_cmd will get no response. 

Below is my csr pskey config and hci_trasnsport_uart_config:

// minimal CSR init script to configure PSKEYs and activate them. It uses store 0x0008 = psram.
static const uint8_t init_script[] = { 

    // 0x01fe: Set ANA_Freq to 26MHz
    0x00, 0xFC, 0x13, 0xc2, 0x02, 0x00, 0x09, 0x00, 0x01, 0x00, 0x03, 0x70, 0x00, 0x00, 0xfe, 0x01, 0x01, 0x00, 0x08, 0x00, 0x90, 0x65,

    // 0x00f2: Set HCI_NOP_DISABLE
    0x00, 0xFC, 0x13, 0xc2, 0x02, 0x00, 0x09, 0x00, 0x01, 0x00, 0x03, 0x70, 0x00, 0x00, 0xf2, 0x00, 0x01, 0x00, 0x08, 0x00, 0x01, 0x00,

    // 0x01bf: Enable RTS/CTS for BCSP (0806 -> 080e)
    0x00, 0xFC, 0x13, 0xc2, 0x02, 0x00, 0x09, 0x00, 0x01, 0x00, 0x03, 0x70, 0x00, 0x00, 0xbf, 0x01, 0x01, 0x00, 0x08, 0x00, 0x0e, 0x08,

    // 0x01ea: Set UART baudrate to 115200
    //0x00, 0xFC, 0x15, 0xc2, 0x02, 0x00, 0x0a, 0x00, 0x02, 0x00, 0x03, 0x70, 0x00, 0x00, 0xea, 0x01, 0x02, 0x00, 0x08, 0x00, 0x01, 0x00, 0x00, 0xc2,

    // 0x01ea: Set UART baudrate to 921600
    0x00, 0xFC, 0x15, 0xc2, 0x02, 0x00, 0x0a, 0x00, 0x02, 0x00, 0x03, 0x70, 0x00, 0x00, 0xea, 0x01, 0x02, 0x00, 0x08, 0x00, 0x0E, 0x00, 0x00, 0x10,

    // 0x0001: Set Bluetooth address 
    0x00, 0xFC, 0x19, 0xc2, 0x02, 0x00, 0x0A, 0x00, 0x03, 0x00, 0x03, 0x70, 0x00, 0x00, 0x01, 0x00, 0x04, 0x00, 0x08, 0x00, 0xf3, 0x00, 0xf5, 0xf4, 0xf2, 0x00, 0xf2, 0xf1,

    //  WarmReset
    0x00, 0xFC, 0x13, 0xc2, 0x02, 0x00, 0x09, 0x00, 0x03, 0x0e, 0x02, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00,
};

static const hci_transport_config_uart_t config = {
    HCI_TRANSPORT_CONFIG_UART,
    115200,
    921600,
    1,
    NULL
};

mringwal, 2022-02-10:
Hi. You should not need to modify the PSKEY init script, as `update_init_script_command` will set the baud rate.
Please try to first use 11520 as main baud rate to see if everything works in general, then try a higher rate, e.g. 230400. Do you use our stm32f4-discovery-cc256x port and habe replaced the chipset driver?
If it doesn't work, please check/post UART traces from a logic analyzer.

xihua13104, 2022-02-10:
Thanks for your reply. 
Yes i use stm32f4-discovery-cc256x port and replaced the chipset driver. The **update_init_script_command** will set the baud rate to baudrate_main, so i think there is no influence that i modified the PSKEY init script. I will try a lower baud rate.

xihua13104, 2023-01-17:
With the help of Logic Analyzer,  I found that this issue was caused by the wrong configuration of STM32F4 clock. It is time to close this issue.

closed 2023-01-17 by xihua13104

mringwal, 2023-01-17:
Glad you found it!