bluekitchen/btstack #408: API function to get BTstack version

electroalper, 2022-03-22:
It would be really nice to have BTstack version info within the source files. 
An API function like `const char *get_btstack_version(void)`would be useful for us to track the library version within our firmware. Using preprocessor define's is also fine.
We regularly update the btstack source code in our project and feel the absence of such a version retrieval function. 
Regards

mringwal, 2022-03-22:
Hi. I like the idea. I'd prefer to have both the release version (e.g. v1.5.3) as well as the git commit hash available. 

However, I don't know how to provide the git version in a source file (there could be a script that fetches the git commit hash, but that would depend on the build system: cmake, make, IDEs, ...)

Is the release version enough for your use case? Any idea how to automatically provide the git commit hash?

electroalper, 2022-03-24:
Hi Matthias,

To give you an idea, on my previous projects, I was calling the following script just before the build operation. (For Linux)
The output is something like `2022_02_15_15_18_bac218e-dirty`. Date and time info belongs to the latest commit message within the git tree. There is also a short hash and an optional "-dirty" suffix to show if the git tree is clean or not.

- I put this output to a header file named `gitversion.h` with a preprocessor definition. 
 `#define GITVERSION "2022_02_15_15_18_bac218e-dirty"` 
- Than I use this string within my project: 
  `printf("Git Version: %s", GITVERSION)`
- The key point with this method is  `gitversion.h` file must be added to the `.gitignore` list.

Script:
```
#!/bin/sh
# Print additional version information for non-release trees.
# Generic script customized for atlas usage

usage() {
	echo "Usage: $0 [srctree]" >&2
	exit 1
}

cd "${1:-.}" || usage

# Check for git and a git repo.
if git rev-parse --verify --short HEAD >/dev/null 2>&1; then

	# Get the last commit date within the current directory
	COMMIT_DATE=$(git log -1 --format="%ci" . | sed -e 's/[- :]/_/g' | cut -b -16)
	printf '%s' $COMMIT_DATE

	# Do we have an untagged version?
	if git name-rev --tags HEAD | grep -E '^HEAD[[:space:]]+(.*~[0-9]*|undefined)$' > /dev/null; then
		COMMIT_ID=$(git log -1 --format="%h" .)
		printf '%s' "_$COMMIT_ID"
	else
		(git describe || git describe --tags || git describe --all --long) \
			2>/dev/null | awk -F- '{printf(" %s", $(NF))}'
	fi

	# Are there uncommitted changes?
	git update-index --refresh --unmerged > /dev/null
	if git diff-index --name-only HEAD . | grep -v "^scripts/package" | read dummy; then
		printf '%s' -dirty
	fi

	# All done with git
	exit
fi
```

electroalper, 2022-03-24:
Later on I switched to a different approach: **Using the git commit count as a version number**.  
Something like `Major.Minor.GitCommitCount`

- Take the commit count using: `git rev-list --count HEAD .`
- Put this number in a preprocessor define in a header file
- Add this header to the project and use previous definition
- Add header file to gitignore list
- Use git pre commit hooks to bring "GitCommitCount + 1" to the beginnig of the new commit message. Something like `"CC1234: ..."`

I don't know if this idea is suitable for multi branch development. For our use case, we have only one master branch and this idea is working great.

mringwal, 2022-03-24:
Thanks for the suggestions. My point was less on how to get the version in a script, but that a) running a script is a pain in one of the ports where a vendor IDE has to be used and b) shell scripts are a bit tricky on Windows.

Nevertheless, most ports use either make or cmake, where running a script is easy (which we use for the Python GATT DB 'compiler'), we could run a script that writes the version into a file which is then picked up by the port specific main.c file and used there.

Btw. we're used this one for the daemon version:
https://github.com/bluekitchen/btstack/blob/master/tool/get_version.sh

And that one just to get the version for the package.sh script:
https://github.com/bluekitchen/btstack/blob/master/tool/get_git_version.sh

I haven't used git hooks in any way,  trying to avoid complexity where possible. I'll have a go at providing a version to unix-style targets (namely libusb and posix-*). For others, I don't want to introduce the risk of build errors for Windows users.

Any idea how to get git version without a shell? I think the ESP32 esp-idf does not have a proper unix-y environment, but I guess git could be run from a Python script (and we mandate Python already).