bluekitchen/btstack #409: Build error without ENABLE_LE_PERIPHERAL macro

kchuva, 2022-03-30:
**Describe the bug**
I use btstack only for BLE Central role, so I undefine `ENABLE_LE_PERIPHERAL` macro in btstack_config. Then compiler generate error:
`error C2039: 'le_advertisements_todo': is not a member of 'hci_stack_t'`
for `hci_le_random_address_set`, `hci_run_general_gap_le` methods.
Because `le_advertisements_todo` member depends on `ENABLE_LE_PERIPHERAL` macro, but those methods implemented without any dependencies.

**To Reproduce**
Steps to reproduce the behavior:
1. Undefine `ENABLE_LE_PERIPHERAL` macro
2. Build project

**Environment:  (please complete the following information):**
 - Current BTstack branch: [e.g. develop]

mringwal, 2022-03-31:
Thanks for reporting. Please try the current develop branch and let me know if it builds for you now.

kchuva, 2022-05-18:
Sorry for late answer. The current branch builds successfully after commit [54da687ebf0496c1b6b273b867ad61e57395ce7b](54da687ebf0496c1b6b273b867ad61e57395ce7b)