bluekitchen/btstack #414: ESP32/develop branch: watchdog triggered in iterating/removing item from list

ricardoquesada, 2022-04-17:
**Describe the bug**

This is hard to reproduce bug, but filing the bug just to raise awareness.

`master` branch seems to be more stable, but in `develop` branch (updated as of today) I started to see watchdogs that are being triggered when removing or iterating a list in btstack.

For example, this watchdog:

```
0x400f2aa7: task_wdt_isr at /home/riq/bin/esp/esp-idf/components/esp_system/task_wdt.c:183 (discriminator 3)
                                                                                                                                            
0x40082c7d: _xt_lowint1 at /home/riq/bin/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111                                                                                                                                                                               
                                                                                                                                            
0x4011ce39: btstack_linked_list_iterator_has_next at /home/riq/progs/bluepad32/src/components/btstack/src/btstack_linked_list.c:162         
                                                                                                                                            
0x400e7498: l2cap_hci_event_handler at /home/riq/progs/bluepad32/src/components/btstack/src/l2cap.c:3028                                                                                                                                                                                 
                                                                                                                                            
0x400dec61: hci_emit_event at /home/riq/progs/bluepad32/src/components/btstack/src/hci.c:6327                      
                                                                      
0x400df17d: hci_emit_l2cap_check_timeout at /home/riq/progs/bluepad32/src/components/btstack/src/hci.c:6476                                 
 (inlined by) hci_connection_timeout_handler at /home/riq/progs/bluepad32/src/components/btstack/src/hci.c:427
                                                                                                                                            
0x400de7ad: btstack_run_loop_base_process_timers at /home/riq/progs/bluepad32/src/components/btstack/src/btstack_run_loop.c:112
                                                                                                                                            
0x400e850a: btstack_run_loop_freertos_execute at /home/riq/progs/bluepad32/src/components/btstack/platform/freertos/btstack_run_loop_freertos.c:187

0x400de8be: btstack_run_loop_execute at /home/riq/progs/bluepad32/src/components/btstack/src/btstack_run_loop.c:293

0x400d675f: uni_main at /home/riq/progs/bluepad32/src/components/bluepad32/uni_main.c:43

0x400d6615: app_main at /home/riq/progs/bluepad32/src/main/main.c:49

0x4011db53: main_task at /home/riq/bin/esp/esp-idf/components/freertos/port/port_common.c:129
```

Which translates to the following:

```
linked_list: btstack_linked_list_iterator_has_next()
l2cap: l2cap_hci_event_handler()
hci: hci_emit_event()
hci: hci_emit_l2cap_check_timeout()
```

And the list iteration seems pretty straight forward... but the watchdog is being triggered in the list. And although I cannot reproduce it in a reliable way, when the watchdog triggers, it is always in the list-iteration or list-deletion code.
It seems that this started to happen more recently in `develop` branch. I wonder if there has been changes in how the list is being handled.

**To Reproduce**

Unfortunately I don't have a good reproduction case... but these crashes seems to appear when at BR/EDR connection time.. incoming connection mostly (?).

**Expected behavior**
Should not crash.

**HCI Packet Logs**

Pending... I'll try to get them.

**Environment:  (please complete the following information):**
 - Current BTstack branch: develop branch, as of today
 - Bluetooth Controller: ESP32
 - Remote device: Switch controller Pro
 - Compiled using release mode
 - Using ESP-IDF v4.4 

**Additional context**

A user of Bluepad32 (which uses btstack) started to see similar crashes when switched from `master` to `develop` ( https://gitlab.com/ricardoquesada/bluepad32/-/issues/12 ).
Unfortunately I cannot reproduce those crashes (which happens on list-deletion), but I can reproduce ones that happens in list-iteration.

mringwal, 2022-04-17:
Hi @ricardoquesada. On ESP32, what does it mean that the Watchdog got triggered? Usually with watchdogs, some functions needs to be called regularly, to know that the system is alive and the watchdog trigger if that doesn't happen.
BTstack's run loop does not trigger anything explicitly. If we have a thread that calculates prime numbers, will that trigger the Watchdog, too?

Ok, assuming this means that BTstack's thread spent all time in a simple linked list iterator. Then, this means that the list is broken, e.g. having become circular. This cannot happen with the existing code - unless linked list functions are called from different contexts, e.g. threads and/or ISRs. If you can reproduce it, please add a counter to the list iterator that counts up to 1000 and if that happens, try to manually print out the list (just to confirm, that it got bad).

There have been no changes to BTstack w.r.t. to its list implementation. That's rather old, simple and well tested code, so I would look deeper into the user code. Another idea would be to add checks to all functions that modify lists and e.g. check a) if you're on a regular thread (not in an ISR), and b) if you're on the BTstack main thead. If one of these two conditions are false, trigger an assert with backtrace. (of course, you can use the list functions from a different thread, but they can only used by a single thread - unless a OS mutex is used).

ricardoquesada, 2022-04-17:
Thanks Matthias. I'll dig deeper... 
I'm 99% that I'm doing everything from the BTStack thread... but I'll double-check everything.

Thanks.

ricardoquesada, 2022-04-17:
QQ: is there a way to know whether a function is running on the btstack main thread?

e.g: something like this:  `assert(btstack_is_running_on_main_thread());`

ricardoquesada, 2022-04-18:
Might be a red-herring, but it seems that most of the watchdogs (not all, just most), have one ancestor function is common which is: `hci_connection_timeout_handler()`

I decided to put a `printf()` in that function to see how often it was called. 
I noticed that under "normal" circumstances, I only gets called once the connection_on_complete cannot be finished... which looks normal.

But in the latest crash, that functioned called multiple times per second, triggering the watchdog.
By looking at the code, that shouldn't happen... but I'm not that familiar with the code, and I wonder if some new logic was recently fixed/added/modified in how HCI handle the connections and the timeout.

Thanks.

mringwal, 2022-04-18:
> QQ: is there a way to know whether a function is running on the btstack main thread?
> 
> e.g: something like this: `assert(btstack_is_running_on_main_thread());`

No, there isn't. In general, it should be clear if code run on main thread or not. If you have some logic that requires action from the stack / on the main thread, you can schedule a callback using btstack_run_loop_execute_on_main_thread - which can be called from any thread (incl. main thread).

mringwal, 2022-04-18:
> Might be a red-herring, but it seems that most of the watchdogs (not all, just most), have one ancestor function is common which is: `hci_connection_timeout_handler()`
>
> I decided to put a `printf()` [...]

Again, no obvious changes to the scheduler. From your description, it looks like a list gets corrupted. Please verify, if that happens or not. Some suggestions above ( a - always check task, b - count iterations in list iterator)

To check the current thread on esp32 with freertos, it looks like you can use this one here https://www.freertos.org/a00021.html#xTaskGetCurrentTaskHandle

ricardoquesada, 2022-04-18:
yes, at one point the list got corrupted... it entered into an infinite loop: it was a one entry list pointing to itself.

Although AFAIK, all my code is running on the btstack thread, I'm not doing allocations/deletions in runtime... everything is pre-allocated at compile time...

The tricky part is that it is very difficult to reproduce... and only happens, apparently with Nintendo Switch controllers... 

I'm still double-checking everything...

mringwal, 2022-04-18:
yupp, that's no fun to debug. From my experience, it's either some general memory corruption (write out of bounds), or, ISR or other thread is modifying a list (without a mutex in place). That's why I'm suggesting to add as many asserts as possible: add a function that's called by all list functions. In that function, check if you're on the btstack thread (assuming btstack  lists are not used by other parts of your code). You can also add functions that counts the number of items in a list and is called before and after each list operation (it should abort if num items > 1000).

mringwal, 2023-02-21:
Closing for now. Feel free to re-open when you have additional information.

closed 2023-02-21 by mringwal