bluekitchen/btstack #415: Cannot Established SPP connection between Android phone

anhpt26, 2022-04-20:
Hi all!
Currently, we use SPP Counter demo to establish connection with android phone and face the issue!
After getting the pairing key for the connection, the android phone sends the disconnection request. 
Please help us to figure out the root cause of this issue!

Here is our spp code demo

`/*
 * Copyright (C) 2014 BlueKitchen GmbH
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions
 * are met:
 *
 * 1. Redistributions of source code must retain the above copyright
 *    notice, this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright
 *    notice, this list of conditions and the following disclaimer in the
 *    documentation and/or other materials provided with the distribution.
 * 3. Neither the name of the copyright holders nor the names of
 *    contributors may be used to endorse or promote products derived
 *    from this software without specific prior written permission.
 * 4. Any redistribution, use, or modification is done solely for
 *    personal benefit and not for any commercial purpose or for
 *    monetary gain.
 *
 * THIS SOFTWARE IS PROVIDED BY BLUEKITCHEN GMBH AND CONTRIBUTORS
 * ``AS IS'' AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 * LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
 * FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL BLUEKITCHEN
 * GMBH OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
 * INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING,
 * BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS
 * OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED
 * AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
 * OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF
 * THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF
 * SUCH DAMAGE.
 *
 * Please inquire about commercial licensing options at 
 * contact@bluekitchen-gmbh.com
 *
 */

#define BTSTACK_FILE__ "spp_counter.c"

// *****************************************************************************
/* EXAMPLE_START(spp_counter): SPP Server - Heartbeat Counter over RFCOMM
 *
 * @text The Serial port profile (SPP) is widely used as it provides a serial
 * port over Bluetooth. The SPP counter example demonstrates how to setup an SPP
 * service, and provide a periodic timer over RFCOMM.   
 *
 * @text Note: To test, please run the spp_counter example, and then pair from 
 * a remote device, and open the Virtual Serial Port.
 */
// *****************************************************************************

#include <inttypes.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
 
#include "btstack.h"
#include "btstack_debug.h"

#define RFCOMM_SERVER_CHANNEL 1
#define HEARTBEAT_PERIOD_MS 1000

[... 716 lines cut ...]
[00:00:02.130] EVT <= 6E 00 
[00:00:02.130] ACL => 0B 00 10 00 0C 00 01 00 04 0A 08 00 50 00 00 00 01 02 97 06 
[00:00:02.130] EVT <= 6E 00 
[00:00:02.133] ACL <= 0B 20 10 00 0C 00 01 00 04 08 08 00 46 00 00 00 01 02 00 04 
[00:00:02.133] ACL => 0B 00 12 00 0E 00 01 00 05 08 0A 00 50 00 00 00 00 00 01 02 00 04 
[00:00:02.134] EVT <= 6E 00 
[00:00:02.134] EVT <= 13 05 01 0B 00 02 00 
[00:00:02.134] ACL <= 0B 20 0E 00 0A 00 01 00 05 0A 06 00 46 00 00 00 00 00 
[00:00:02.134] EVT <= 70 18 00 7E 10 7D A0 BF 08 0B 00 01 00 46 00 50 00 97 06 00 04 00 00 01 00 01 
[00:00:02.137] ACL <= 0B 20 18 00 14 00 46 00 06 00 00 00 0F 35 03 19 01 00 03 F0 35 05 0A 00 00 FF FF 00 
[00:00:02.138] EVT <= 79 02 46 00 
[00:00:02.138] ACL => 0B 00 6E 00 6A 00 50 00 07 00 00 00 65 00 62 36 00 5F 36 00 5C 09 00 00 0A 00 01 00 01 09 00 01 36 00 03 19 11 01 09 00 04 36 00 0E 36 00 03 19 01 00 36 00 05 19 00 03 08 01 09 00 05 36 00 03 19 10 02 09 00 06 36 00 09 09 65 6E 09 00 6A 09 01 00 09 00 09 36 00 09 36 00 06 19 11 01 09 11 02 09 01 00 25 0B 53 50 50 20 43 6F 75 6E 74 65 72 00 
[00:00:02.139] EVT <= 6E 00 
[00:00:02.141] EVT <= 13 05 01 0B 00 01 00 
[00:00:02.141] ACL <= 0B 20 0C 00 08 00 01 00 06 09 04 00 46 00 50 00 
[00:00:02.142] ACL => 0B 00 0C 00 08 00 01 00 07 09 04 00 46 00 50 00 
[00:00:02.142] EVT <= 71 02 46 00 
[00:00:02.142] EVT <= 6E 00 
[00:00:02.144] EVT <= 13 05 01 0B 00 02 00 
[00:00:02.269] EVT <= 05 04 00 0B 00 13 
[00:00:02.269] EVT <= 61 01 00 
`

Best Regards
Pham Tuan Anh

anhpt26, 2022-04-21:
Hi Mringwal!

Thank you so much for your response!

My BT set-up is a microcontroller communicating with a Cypress module.

I tried to apply your recommendation to my source code. But right now, the situation is the same as before. My BT setup board cannot connect with the Android phone.

Here are my bt-snooping log and btstack_config.c. The bt snooping log is recorded from I set-up SPP service to the BT set-up receiving the disconnect request from the Android phone.
[https://drive.google.com/drive/folders/15QWFI4qdW1kxIch-KpoDz68PzbBFxUcq?usp=sharing](url)

Please help me to figure out my mistake!

mringwal, 2022-04-21:
Hi again. The btstack_config.h mentions an ez430-cc256x, what microcontroller do you use?

Anyway, the Cross-transport key derivation looks good. I've seen that Android uses "dedicated bonding". In dedicated bonding, the initiating device disconnects after the pairing is complete. So, there's nothing really wrong here.

Could you try a different SPP Terminal app on Android?

anhpt26, 2022-04-22:
Hi Mringwal!

Thanks for your support!
Currently , my BT set-up can connect with the Android phone!
The issue come from SPP terminal app on my phone. It has the issue when connect with my BT set-up.
After changing new SPP terminal app, I can establish the connection between my phone and BT set-up.

Thank you again and sorry for this inconvenience!

Best Regards
Pham Tuan Anh

anhpt26, 2022-04-22:
Hi Mringwal!
I would like to close this ticket!
Thanks for your support

closed 2022-04-22 by anhpt26