bluekitchen/btstack #422: hci: Allow to store link keys even if the "auth_req" is "not required"

ricardoquesada, 2022-05-25:
**Is your feature request related to a problem? Please describe.**

Some gamepads, in particular Nintendo Switch controllers (JoyCons and Switch Pro), when they establish a connection, they have a `conn->io_cap_response_auth_req = SSP_IO_AUTHREQ_MITM_PROTECTION_NOT_REQUIRED_NO_BONDING`

But that doesn't mean that the link key must not be stored.
In fact, these controllers support "reconnect", but in order to work, the link key must be stored.

**Describe the solution you'd like**

The problem seems to be in hci.c, in particular in line 3184:
https://github.com/bluekitchen/btstack/blob/master/src/hci.c#L3184

```c
            if (conn->link_key_type != COMBINATION_KEY){
                bool remote_bonding = conn->io_cap_response_auth_req >= SSP_IO_AUTHREQ_MITM_PROTECTION_NOT_REQUIRED_DEDICATED_BONDING;
                if (!remote_bonding){
                    break;
                }
            }
```

**Describe alternatives you've considered**

First of all, I don't understand why the Nintendo Switch gamepads set the "no bonding" whey they do support re-connect.

But ideally, the link key should be stored, even when "no bonding" is specified in "Io capabilities".

The specification says (From Bluetooth v4.2, Chapter 6.5.3.1 General Bonding):

> 'No bonding' is used when the device is performing a Secure Simple Pairing procedure, but does not intend to retain the link key after the physical link is disconnected.

But it doesn't say that the link key should not be stored...

Alternative: Add an API to force storing link-keys even when "io capabilities" says "no bonding" (?)

**Additional context**
Add any other context or screenshots about the feature request here.

ricardoquesada, 2022-05-26:
I did the same experiment in Linux + Bluez. I captured the Bluetooth packets, and I confirm that Nintendo Switch controller is sending the "no bonding" auth_req, but Bluez saves it.

I rebooted my Linux machine just to make sure that no "cache" was saved.
And I confirm that the link key was still stored somewhere in BlueZ and I was able to reconnect the Switch controller on Linux.

I'm not familiar with the BlueZ code, and I tried to find where it is storing the link-key, but I failed... (although I haven't tried that hard).

mringwal, 2023-11-27:
Hi @ricardoquesada Could you check if BTstack develop branch is storing link key although remote does specify "no bonding", e.g. by repairing to your Nintendo Switch?

mringwal, 2024-10-14:
Hi @ricardoquesada 

Could you post a HCI log for pairing with the Nintendo Switch Controller?

I'm having an issue with another customer that is irritated about link keys getting stored although both sides did set "no bonding", which is probably wrong. I'd like to see if there's compromise possible to handle both situations without a compile time flag or API to force storing the keys. They are in Acceptor role while I assume you're in Initiator role.

ricardoquesada, 2024-10-14:
sure. tonight or tomorrow I'll post the HCI dump

ricardoquesada, 2024-10-15:
[hci_dump.zip](https://github.com/user-attachments/files/17370732/hci_dump.zip)

dump from a Nintendo Switch Controller connected to a bluepad32 + btstack + Linux

mringwal, 2024-11-10:
FWIW the code on develop will now store the link key if BTstack did initiate the authentication (usually also the same as when creating a new connection). If the other side starts the authentication and does not require bonding, BTstack will also send "no bonding" and not store the link key in the end. So, it shold both for the Nintendo switch and it also works for temporary pairing initiated by the other side.