bluekitchen/btstack #425: Esp32-c3

mickeyl, 2022-05-30:
Would this library work on the new riscV based esp32-variant Esp32-c3?

mringwal, 2022-05-30:
Hi @mickeyl - in theory it should work - there's no obvious reason that would prevent it and the VHCI interface hasn't changed from the original ESP32 - but it didn't start up one the first try I did a while ago. 

Getting it to work is on my list, but I didn't get around to look a bit more in detail.
It's possible to use the C3 as an external Bluetooth Controller connected via UART - I've successfully tried that recently.

mringwal, 2022-06-08:
Hi again. Turns out, we had to swap two lines to get it working (and sort out other differences, e.g. due to the I2S GPIOs not existing on the smaller ESP32-C3).

Anyway, please get the latest version on the develop branch and follow the port/esp32/README.md to get started.
Btw.. I still have an SIMpad in the basement... :)

KingMichaelCoy, 2022-09-07:
I also have a problem. My ArduinoCode code uploads successfully of a ble controller sketch but Bluetooth does not show on laptop or cellphone as midi device? What do I do?

mringwal, 2022-09-08:
@KingMichaelCoy We don't support the Arduino environment with BTstack. Please post on the Espressif's Arduino forum/project.

(Closing this issue as ESP32-C3 is supported on develop and we'll mrege onto master soon)

closed 2022-09-08 by mringwal