bluekitchen/btstack #428: sco_data_packet_length, info request (maybe bug?)

amorniroli, 2022-06-07:
Hi @mringwal,

just a question about SCO packet length
```
#ifdef ENABLE_SCO_OVER_HCI
    // Transparent = mSBC => 1, CVSD with 16-bit samples requires twice as much bytes
    int multiplier = ((hci_stack->sco_voice_setting_active & 0x03) == 0x03) ? 1 : 2;

    if (hci_have_usb_transport()){
        // see Core Spec for H2 USB Transfer. 
        // 3 byte SCO header + 24 bytes per connection
        int num_sco_connections = btstack_max(1, hci_number_sco_connections());
        sco_packet_length = 3 + 24 * num_sco_connections * multiplier;
    } else {
        // 3 byte SCO header + SCO packet size over the air (60 bytes)
        sco_packet_length = 3 + 60 * multiplier;
        // assert that it still fits inside an SCO buffer
        if (sco_packet_length > hci_stack->sco_data_packet_length){
            sco_packet_length = 3 + 60;
        }
    }
#endif
```

In my case, **_hci_stack->sco_data_packet_length_** is 120bytes and **_sco_packet_length_** is 123 bytes. Shouldn't be the assert check be done in the following way, since sco_data_packet_length  is the payload size?

```
// assert that it still fits inside an SCO buffer
if (sco_packet_length > (hci_stack->sco_data_packet_length + 3)){
    sco_packet_length = 3 + 60;
}
```

Otherwise sco_packet_length will always be truncated to 63.

mringwal, 2022-06-07:
Hi @amorniroli 

Thanks for pointing this out. Yes, the check misses the 3 SCO header bytes to be correct. 
What Bluetooth Controller is this? I think the ones I've tried did return a buffer larger than 120 bytes and then the if clause wasn't entered.

In any case, the demo code would just just a smaller SCO packet and the Controller should combine them again.

I've updated the code similar to your suggestion on develop.
I'm unclear if there's a situation / Controller where this would be needed,

closed 2022-06-07 by mringwal

amorniroli, 2022-06-07:
Hi @mringwal 

the controller is always the CC2564 (still waiting for Infineon/Laird EVBs :D).

There is not situation, it was just a clarification in order to "optimize" TX in just a single packet.

Thanks for the support.