bluekitchen/btstack #429: Connection updates can interrupt service/characteristic discovery and lead to abrupt disconnect

dpgeorge, 2022-07-13:
### Description

We are seeing intermittent issues when using `gap_request_connection_parameter_update()` (on the peripheral side).  The central sometimes gives an error like this:
```
[01:39:47.221] LOG -- hci.c.2123: hci_number_completed_packet lists unused con handle 64
```

### Set up
Peripheral side setup: a simple advertising peripheral that can be connected to and has a service with two characteristics.  When the connection event comes through it immediately requests to update the connection parameters by calling `gap_request_connection_parameter_update(conn_handle, 0xc, 0xc, 0, 0x48);`.

Central side setup: scans for the peripheral, connects to it, discovers all services, then discovers all characteristics for the found service (there is only one service on the peripheral).  Then it waits 3 seconds and disconnects.

Note that everything here is running MicroPython+BTstack.  I have not yet tried to reproduce the problem using pure BTstack.

### Behaviour

A single run is the central making a connection, doing discovery, and then disconnecting.  This is done over and over again.  The central code prints out BLE events as they occur.

A good run has the following output:
```
Found peripheral
peripheral connect 64
service UUID(0x1800)
service UUID(0x1801)
service UUID(0xfd02)
service done
characteristic UUID('0000fd02-0001-1000-8000-000000000000')
characteristic UUID('0000fd02-0002-1000-8000-000000000000')
characteristic done
connection_update (64, 12, 0, 72, 0)
peripheral disconnect 64
```

A failed run looks like this (notice that the connection_update event comes in before the end of service discovery):
```
Found peripheral
peripheral connect 64
service UUID(0x1800)
service UUID(0x1801)
service UUID(0xfd02)
connection_update (64, 12, 0, 72, 0)
service done
peripheral disconnect 64
[01:39:47.221] LOG -- hci.c.2123: hci_number_completed_packet lists unused con handle 64
```

A different failure mode is when connection_update happens during characteristic discovery:
```
Found peripheral
peripheral connect 64
service UUID(0x1800)
service UUID(0x1801)
service UUID(0xfd02)
service done
connection_update (64, 12, 0, 72, 0)
characteristic done
Failed to find characteristic.
peripheral disconnect 64
```

In all failures the peripheral disconnects abruptly.  In the success case the disconnect is initiated by the central after 3 seconds of being connected.

### Notes

- BTstack is running on both central and peripheral.  No issue is seen when running NimBLE on the central instead.
- BT chipset is CYW4343 on the central and CC2564 on the peripheral.  No issue is seen when the peripheral chipset is a CYW4343.  Issue is still seen when both sides are CC2564.
- It looks like a timing sensitive problem, possibly related to the time that the connection update request is sent.

I know this might not be enough info to debug, so let me know what else I can do to facilitate debugging.  I can provide full code for the above, and/or binary firmware.

mringwal, 2022-07-13:
Hi @dpgeorge Please provide HCI dumps incl. log_info at least for the Central side to see whats going on/failing. Thx!

dpgeorge, 2022-07-13:
I only have the MicroPython-level HCI dumps easily available (BTstack dump is a bit more effort, but can do it if needed).  Here is a failed case (< means out to chipset, > means in to the MCU, number in brackets is timestamp in milliseconds):
```
< [  137283] 01:0b:20:07:00:30:00:30:00:00:00
> [  137284] 04
> [  137284] 0e:04
> [  137284] 01:0b:20:00
< [  137284] 01:0c:20:02:01:00
> [  137285] 04
> [  137285] 0e:04
> [  137285] 01:0c:20:00
> [  137293] 04
> [  137293] 3e:29
> [  137293] 02:01:00:00:fd:ef:b9:49:e7:e6:1d:02:01:06:19:16:f7:fd:01:66:f9:d4:84:08:3c:57:2b:4d:7a:34:14:3f:7a:46:ba:00:00:00:00:03:a6
> [  137305] 04
> [  137305] 3e:2b
> [  137305] 02:01:03:01:d3:96:7d:6f:d7:38:1f:1e:ff:06:00:01:09:20:02:52:a8:a8:8e:4f:35:23:d7:33:dd:a5:a7:0d:f8:52:c9:12:eb:84:f5:6d:bf:cf:a0
> [  137326] 04
> [  137326] 3e:2b
> [  137326] 02:01:03:01:0d:88:7f:1b:da:27:1f:1e:ff:06:00:01:09:20:02:f1:b6:bf:c5:12:ea:ad:26:17:6d:44:f9:e1:ab:d8:a8:82:46:60:7f:34:41:c8:a1
> [  137339] 04
> [  137339] 3e:28
> [  137339] 02:01:02:01:30:6c:4a:be:be:7e:1c:03:03:9f:fe:17:16:9f:fe:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:a0
> [  137346] 04
> [  137346] 3e:13
> [  137346] 02:01:00:00:df:21:ab:3c:0b:38:07:02:01:06:03:03:02:fd:dd
< [  137347] 01:0c:20:02:00:00
Found peripheral
> [  137348] 04
> [  137348] 0e:04
> [  137348] 01:0c:20:00
< [  137348] 01:0d:20:19:60:00:30:00:00:00:df:21:ab:3c:0b:38:00:08:00:18:00:04:00:c8:00:10:00:30:00
> [  137349] 04
> [  137350] 0f:04
> [  137350] 00:01:0d:20
> [  137504] 04
> [  137504] 3e:13
> [  137504] 01:00:40:00:00:00:df:21:ab:3c:0b:38:18:00:04:00:c8:00:00
peripheral connect 64
< [  137504] 02:40:00:0b:00:07:00:04:00:10:01:00:ff:ff:00:28
> [  137562] 02
> [  137562] 40:20:10:00
> [  137562] 0c:00:05:00:12:01:08:00:0c:00:0c:00:00:00:48:00
< [  137562] 02:40:00:0a:00:06:00:05:00:13:01:02:00:00:00
< [  137592] 01:13:20:0e:40:00:0c:00:0c:00:00:00:48:00:00:00:ff:ff
> [  137592] 04
> [  137592] 13:05
> [  137592] 01:40:00:02:00
> [  137593] 04
> [  137593] 0f:04
> [  137593] 00:01:13:20
> [  137622] 02
> [  137622] 40:20:18:00
> [  137622] 14:00:04:00:11:06:01:00:03:00:00:18:04:00:06:00:01:18:07:00:0c:00:02:fd
service UUID(0x1800)
service UUID(0x1801)
service UUID(0xfd02)
< [  137623] 02:40:00:0b:00:07:00:04:00:10:0d:00:ff:ff:00:28
> [  137891] 04
> [  137891] 13:05
> [  137891] 01:40:00:01:00
> [  137936] 04
> [  137936] 3e:0a
> [  137936] 03:00:40:00:0c:00:00:00:48:00
connection_update (64, 12, 0, 72, 0)
> [  138657] 04
> [  138657] 05:04
> [  138657] 00:40:00:08
service done
< [  138657] 02:40:00:0b:00:07:00:04:00:08:07:00:0c:00:03:28
peripheral disconnect 64
> [  138891] 04
> [  138891] 13:05
> [  138891] 01:40:00:01:00
[00:02:18.891] LOG -- hci.c.2123: hci_number_completed_packet lists unused con handle 64
```

dpgeorge, 2022-07-20:
Attached here is the full HCI dump in pklg format.  It includes about 70 of the above events (not all have the issue though).

I'm using BTstack at commit c8b9823f68c6af0fa52e2c4e009aba4dbf257232 which is rather old.  We do want to update to a newer version but that requires some effort.

(Also, the instructions in `.github/ISSUE_TEMPLATE/bug_report.md` mention enabling `LOG_INFO`, but I think it should say `ENABLE_LOG_INFO`.)

[dump.pklg.gz](https://github.com/bluekitchen/btstack/files/9145858/dump.pklg.gz)

dpgeorge, 2022-07-20:
> I would suggest to try slave latency = 4 to see if that avoids the disconnects.

Using latency = 4 still has the same problem.

mringwal, 2022-07-20:
Pity. It's possible to use an LE sniffer, either a cheap one like our open source Raccoon, or a commercial one. However, I would expect the CC2564C to loose the link. In that case, we'd have proof, but I'm pessimistic about the potential to get a fix from TI eventually.

Just curious: do you get the disconnect even without the GATT Client requests?

dpgeorge, 2022-07-20:
Some progress here.  The central (when running BTstack) is connecting with default parameters, which has `conn_latency` at 4.  Changing that to 0 makes the problem go away.

And testing with NimBLE, it had a default value for the connection `conn_latency` of 0, when I didn't see the issue.  Changing it to 4 makes NimBLE also have the same issue as BTstack (although it shows up slightly differently, I think BTstack handles the problem more gracefully).

Summary: when the central initiates the connection with a `conn_latency` of 4, the problem is there.  Using 0, the problem is not there (connection update requests come in and are handled just fine, even if they are interleaved with characteristic scans).

Related to this, I see there are no examples or tests in BTstack for `gap_set_connection_parameters()`.  Also, why is the default for `conn_latency` 4?

mringwal, 2022-07-20:
> Some progress here. The central (when running BTstack) is connecting with default parameters, which has `conn_latency` at 4. Changing that to 0 makes the problem go away.

Well, that's seems like a reasonable workaround for the bug in one of the link layers. (I would assume it's the CC2564C, but we cannot rule out that it's in the 4343 but that they are 'symmetric' in a way that the link loss does not happen with them).

> And testing with NimBLE, it had a default value for the connection `conn_latency` of 0, when I didn't see the issue. Changing it to 4 makes NimBLE also have the same issue as BTstack (although it shows up slightly differently, I think BTstack handles the problem more gracefully).

The stack isn't doing much here. If the link is gone, it's gone. Great that the link stability is independent from the stack.
 
> Summary: when the central initiates the connection with a `conn_latency` of 4, the problem is there. Using 0, the problem is not there (connection update requests come in and are handled just fine, [...]
The `gatt_streamer.c` (Peripheral role) uses gap_request_connection_parameter_update. 

> Also, why is the default for `conn_latency` 4?
conn_latency > 0 should save energy on the Peripheral. Otherwisse, when I didn't have a clear opinion, I've used the params from iOS... (but allowed the user to change them of course)

dpgeorge, 2022-07-21:
Going back to `conn_latency` of 4, I tested with pairs of CC2564 and 4343 chipsets.  This is the result:

| central | peripheral | has problem? |
|-|-|-|
|4343 | 4343 | no|
|CC2564|4343|no|
|4343|CC2564|yes|
|CC2564|CC2564|yes|

Everything was running BTstack+MicroPython.  It really does point to an issue specific to the CC2564.

@thyrrestrup FYI

mringwal, 2022-07-21:
Staying with low-tech options, a third controller, e.g. an USB Bluetooth Dongle would allow to tip the scale one way or the other. My bet would be on the CC2564C, though.

dpgeorge, 2022-08-10:
I tested with a STM32WB55 and updated the table above.  Seems to be pretty conclusive that the CC2564 is the problem.

I think we can close this issue though, because it's not a problem related to BTstack.

closed 2022-08-12 by dpgeorge