bluekitchen/btstack #431: Will the next version support ios?

HiJackhem, 2022-08-25:
Hi,
    The btstack support ios in V1.4.1, but drop ios support in V1.5.0. I want to connect with ios by btstack, will the next version support ios?

Best Regards

mringwal, 2022-08-25:
Hi. Sorry, BTstack won't support iOS in future versions. Cheers.

closed 2022-08-25 by mringwal

HiJackhem, 2022-08-26:
Hi Matthias,

	Thanks for your reply.

	I may not have made it clear what I meant. I have ported btstack with posix-h4, it can pair and connect with android phone and it works fine when play music, make a call.
	
	But it can't work normally with Iphone.When connect with iphone by A2DP_SINK_DEMO, it pairs successfully, but can't connect. When connect with Iphone by HFP_HF_DEMO, it can pair and connect,but no audio data when make a call.

Best Regards

mringwal, 2022-08-26:
I see. HFP HF and A2DP Sink work fine with iOS devices in general. Please describe a concrete example where it doesn't work for you and upload the hci_dump.pklg file for a quick check.

HiJackhem, 2022-08-26:
Hi Matthias,

	Thanks for your patient reply.
	
	After reboot the linux virtual machine, the A2DP_SINK_DEMO works normally with iphone, But HFP_HF_DEMO still has no audio data when making a call.
	
	Below is my develop environment:
	
	linux virtual machine: Ubuntu 22.04.1
	Btstack version: latest version V1.5.3
	BT controller:   esp32，core spec: V4.2
	BT connected source: Iphone 8 or ipad mini 6
	
	The attached files are hfp log for android and Iphone , you can compare them.

Looking forward to your reply.

Best Regards.

[android_hfp_normal.log](https://github.com/bluekitchen/btstack/files/9430924/android_hfp_normal.log)
[ios_hfp_no_audio.log](https://github.com/bluekitchen/btstack/files/9430926/ios_hfp_no_audio.log)

mringwal, 2022-08-26:
Ah.. that's a totally different topic. Thanks for the logs. So, just to confirm: you have the exact same setup (linux vm, btstack, esp32) with iOS and Android device, but with iOS you don't receive audio?

In the logs, everything is as expect, just that a) you don't receive SCO packets from iOS and b) you don't get a Number Complete Events for the SCO packets that you sent. I'm puzzled about this.

As a comparison: could you try the same with port/libusb and a USB Bluetooth dongle and check if it works with both smartphone types?

HiJackhem, 2022-08-26:
Hi Matthias,

> Ah.. that's a totally different topic. Thanks for the logs. So, just to confirm: you have the exact same setup (linux vm, btstack, esp32) with iOS and Android device, but with iOS you don't receive audio?

>###  _yes._

> In the logs, everything is as expect, just that a) you don't receive SCO packets from iOS and b) you don't get a Number Complete Events for the SCO packets that you sent. I'm puzzled about this.

> ### I also find the difference.  May be compatibility problems between controller and host. I will use usb bluetooth dongle for testing.

> As a comparison: could you try the same with port/libusb and a USB Bluetooth dongle and check if it works with both smartphone types?

> ### I will try this. Thanks for your suggestion. 

Best Regards

HiJackhem, 2022-08-29:
Hi Matthias,

	I have tried usb bluetooth dongle with port\libusb. A2DP_DEMO and HFP_DEMO for both android and iphone work fine. Maybe esp32 controller is not compatible with btstack.
	
	Thank you very much.

Best Regards

mringwal, 2022-08-29:
OK, that's good. I don't think it's a problem between BTstack and ESP32, it more looks like a problem between ESP32 and IOS.

What firmware do you currently run on the ESP32 to enable it as an HCI Controller?

Could you try the hfp_ag_demo on the ESP32 directly? see port/esp32
If you enable the sine playback in example/sco_demo_util.c, you don't need to configure audio input/output.
Others have used ESP32 with HFP and iOS, so I assume it works in general. I'd like to see if the difference between BTstack on the ESP32 vs BTstack on another MCU makes a difference (still, I'm puzzled why this could be relevant).

HiJackhem, 2022-09-01:
Hi Matthias,

	I add below MICRO command in /port/posix-h4/btstack_config.h. It resolves the problem. Now iphone hfp works fine with esp32 controller.
	
	#define ENABLE_CLASSIC_LEGACY_CONNECTIONS_FOR_SCO_DEMOS
	
Best Regards

mringwal, 2022-09-02:
HI. Glad that you found this one and I now remember: the ESP32 cannot use SCO when BR/EDR Secure Connections are used. In your case, your iPhone supports BR/EDR Secure Connections, while the Android one doesn't. If you set ENABLE_CLASSIC_LEGACY_CONNECTIONS_FOR_SCO_DEMOS, the code in sco_demo_util.c just disable support for BR/EDR Secure connections.

It would be good to do this automatically in the posix-h4 and windows-h4 ports if the ESP32 is detected as Controller.
I'll make add a ToDo.