bluekitchen/btstack #432: Many errors when compiling Getting Started projects due to type mismatch

jotkali, 2022-10-02:
**Describe the bug**

Unable to compile any sample projects with ESP32 examples due mostly to type mismatch 

**To Reproduce**

1. run ./integrate_btstack.py to install BTStack library and examples into ESP-IDF2. 
2. in port/esp32/ attempt to compile suggested starter project spp_counter with idf.py build3. 
It fails the build with mostly type mismatch errors
(I have listed a small selection of build issues at the end)

**Expected behavior**

Project compiles and is able to be flashed to a ESP32 board

**Environment:  (please complete the following information):**
 - Current BTstack branch: master 1.5.3
 - Bluetooth Controller: ESP32 Adafruit Huzzah 32
 - Remote device: MacBook Air M2 connected via USB to the ESP32 board

**Additional context**
(Please note the ESP-IDF examples do compile idf build, the BTStack examples do not compile)

Errors shown (there are 100s of these type mismatches):

btstack/src/btstack_debug.h:109:70: error: format '%u' expects argument of type 'unsigned int', but argument 5 has type 'uint32_t' {aka 'long unsigned int'}

btstack/src/mesh/mesh.c:1153:66: error: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'uint32_t' {aka 'long unsigned int'}

mesh_configuration_server.c:2021:42: error: format '%d' expects argument of type 'int', but argument 2 has type 'int32_t' {aka 'long int'}

jotkali, 2022-10-04:
Appreciate the response. I downgraded to esp-idf4.4, reinstalled BTStack and the examples compiled. Thanks

mringwal, 2022-10-04:
I'm glad you can get started and I appreciate the warning, as we're not able to test on all platforms continuously. I think IDF 5.0 isn't out yet, let's see how much of these warnings/errors we can get rid of until then.

jotkali, 2022-10-05:
Spoke too soon, under ESP-IDF 4 the example avrcp_browsing_client fails to compile with:

undefined reference to `l2cap_ertm_create_channel'
in
avrcp_retry_timer_timeout_handler
avrcp_browsing_handle_sdp_client_query_result
avrcp_browsing_connect

AVRCP with volume control, forward, reverse, play pause is what brought me to BTStack. Can I help with fixing it? Perhaps point me in the right direction?

mringwal, 2022-10-05:
Hi. Good point. AVRCP Browsing requires L2CAP ERTM. I've updated the develop branch to enable it in btstack_config.h. Please try again.

jotkali, 2022-10-06:
Thx

It builds! it gives one error but easy to fix by initializing the variable l2cap.c function l2cap_ertm_send line 466:
`int chunk_len = 0`

closed 2022-10-10 by mringwal