bluekitchen/btstack #433: Weird connection/disconnection behaviour

fullmetalfelix, 2022-10-04:
I have a raspberry pi zero acting as HID joystick using a program running the btstack (modified HID mouse example), and a windows 10 computer that the pi has to connect to.
I start the program on the pi, btstack goes online, and the computer can find the HID device and connect to it without trouble. The pi is correctly seen as a joystick, and it sends correct HID reports to the computer.

Eventually I kill the program on the pi, but windows 10 still shows the HID device as connected. Even when pi is completely off, cut from the power supply, windows still thinks it is connected. It even appears in the list of USB joysticks, although it does not show any sign of life.

When I start the pi and the program again, it shows the message that btstack is online, but it does not connect to the computer. In order to connect it again, I have to manually disconnect the device from the windows bluetooth control panel, then search for it again, at which point it connects and works as expected.

I would have expected windows to realize the HID joystick was disconnected but paired, and automatically reconnect when the btstack program starts on the pi.
Did I miss something or did something wrong?

Thanks for your help.

mringwal, 2022-10-04:
Hi. BTstack stores the bonding information on Linux systems in /tmp, which gets deleted on a power cycle.
Could you either move it to persistent storage (check TLV_DB_PATH_PREFIX in main.c) or instead of quitting the app and shutting down, just disconnect the connection to Windows by calling gap_disconnect with the connection handle from HID_SUBEVENT_CONNECTION_OPENED.

fullmetalfelix, 2022-10-04:
Hi,
thanks a lot for the swift response. I moved the bonding info to a persistent path. Now when the program on the pi is terminated, windows displays it as "Paired" as I would expect.
However, when the program is restarted and the BTstack goes online, it does not automatically connect.
I tried calling hid_device_connect(addr, &hid_cid) using the address of the pi itself, but fails to connect with status 0xf.
How would I get it to automatically reconnect on start?

mringwal, 2022-10-04:
if you want to connect in a BTstack app running on the Raspi to the Windows PC, you need to specify the Bluetooth Address (BD_ADDR) of the PC in the hid_device_connect call.

Also, please have a look at /tmp/hci_dump.pklg with Wireshark. It is often helpful to just follow the HCI Commands and check the responses.

fullmetalfelix, 2022-10-04:
That is what I would have expected, but how can I get the host address? My guess would be to read it from some packet of some type during the connection procedure, save it in a file, and then try to reconnect to previously seen hosts until one responds.

mringwal, 2022-10-04:
You can get it during connection from HID_SUBEVENT_CONNECTION_OPENED with hid_subevent_connection_opened_get_bd_addr. For a first test, you can write it down and hard-code it.

In practice, you could store the address in BTstack's persistent store via btstack_tlv.h.

fullmetalfelix, 2022-10-04:
I managed to get the server address from the connection_openened event, and it corresponds to the one found in the hci_dump.
This was saved to a file, and loaded when the program restarted.
I setup a separate pthread that periodically checks if the app_status is not connected, and calls hid_device_connect with the server address, but nothing happens.
Any suggestions?

mringwal, 2022-10-05:
BTstack is not thread-safe. You can use BTstack's timer sources, see led_counter.c, to trigger periodic action.
(You could also schedule a callback from the BTstack thread to your code).
You could try to re-connect on startup, and set a timer when it fails, or if you get disconencted.
Please send another log if it doesn't work with the BTstack time.

fullmetalfelix, 2022-10-05:
I implemented it like in the led_counter example but still not working.
Now the btstack_main sets up the heartbeat with 10s timeout, and the heartbeat handler calls hid_device_connect if the app_status is not connected. Before finishing heartbeat handler resets and adds the timer to the main loop (30s delay this time).

hci dump log here: https://www.dropbox.com/s/2sdkx0hdcq2gszr/hci_dump_heartbeat.pklg?dl=0

As expected, after 10s I see the connection attempt and after another 15s it prints "Connection failed, status 0x4"

mringwal, 2022-10-05:
Looks good. if you get statux 0x04, that's a page timeout. Usually, the BD_ADDR is not correct or the device is not in range. Please double check the BD_ADDR. Your code tries to connected to 08:78:A7:B6:08:78.

fullmetalfelix, 2022-10-05:
OK, sorry to have bothered you!
For some reason the address of the bluetooth dongle on the windows computer changed since yesterday. Obviously, the stored address was not good. Now calling hid_device_connect does indeed reconnect the pi to the computer.

Thanks a lot for your help.

mringwal, 2022-10-05:
Glad you got it working

mringwal, 2022-10-05:
well. what about setting the timer on the connection failure?

closed 2022-10-05 by mringwal

fullmetalfelix, 2022-10-05:
Ah yes, event HID_SUBEVENT_CONNECTION_OPENED fires also when it fails. Thanks.