bluekitchen/btstack #434: [regression]: Latest master-branch failed to initialize HCI on ESP32 when BT BR/ERD is enabled (BLE disabled)

ricardoquesada, 2022-10-08:
**Describe the bug**

It used to be possible to use BTstack on ESP32  having only BR/EDR enabled.
With latest master (hash 1635e36d06821af8b61302509e91bfcc1ade84c4), that is no longer the case.

The `BTSTACK_EVENT_POWERON_FAILED` message is received if only BR/EDR is enabled.

**To Reproduce**

Steps to reproduce the behavior:

1. Go to the `port/esp32/example/hid_host_demo` folder.
2. run `idf.py menuconfig`
3. Select: `Component config` -> `Bluetooth` -> `Bluetooth controller` -> `BR/EDR only`
4. Run example 

It won't finish initialization. Instead of receiving a `BTSTACK_EVENT_STATE` with `HCI_STATE_WORKING` it receives a `BTSTACK_EVENT_POWERON_FAILED`.

**Expected behavior**

It should work. Previous master-branch worked Ok.

mringwal, 2022-10-10:
Thanks for reporting. It was a simple typo, fixed on develop. Please verify.

ricardoquesada, 2022-10-11:
seems to be working ok. thanks for the quick fix.

closed 2022-10-11 by ricardoquesada