bluekitchen/btstack #435: Is btstack support multi-threads?Is there any way let my statck connect with multi BT/BLE devices?

SeanPcWoo, 2022-10-14:
Now I run the btstack in my product. The funtions of the product like a PC. It can connect with hid-keyboard 、mouse and audio device.
However I can only run one demo once, it means my product can only conntect  with one device once.
How can I let my product connected with more than one device? I need hid-keyboard and audio device connect to btstack and work.

mringwal, 2022-10-14:
Hi Sean. BTstack allows to use different profiles at the same time. For this, you need to combine both the setup logic as well as the packet handlers, e.g. from different examples. Please see our documentation.

As an example, gatt_counter (GATT Server) and spp_counter (SPP Server) have been combined into spp_and_gatt_counter.

For commercial use, please contact us at contact@bluekitchen-gmbh.com as well.

closed 2022-10-14 by mringwal

SeanPcWoo, 2022-10-17:
Ok, i got it. Thanks!