bluekitchen/btstack #437: Two consecutive HCI_EVENT_CONNECTION_COMPLETE may cause assertion failure

AminoACID123, 2022-10-28:
Two consecutive `HCI_EVENT_CONNECTION_COMPLETE` with the same `BD_ADDR` may cause assertion failure in `btstack_run_loop_add_timer`.  
I understand this might not even be a problem, since a compliant controller won't generate such events. However, in some cases where the controller is a USB dongle plugged in by a malicious attacker, this issue may be exploited to crash the application.  
Perhaps there is a better way dealing with this other than using assertion?

mringwal, 2022-11-01:
Thanks for reporting this. I think there are a number of different angles to look at this.
--
How did you stumble into this?
--
Yes, it would be nice/better if the stack would ignore the second connection complete event as there's already a connection and the stack cannot do anything about it. I'll have a look if this can be achieved with minimal changes
--
In general, a stack cannot provide any functional guarantees if the Controller is malicious, e.g. the Controller could completely ignore the stack as one example. Also, if an attacker is physically present, it's usually game over anyway. Our goal with BTstack is to prevent code execution triggered by a malicious Controller by fully validation all input from the controller.
--
FWIW, I'm currently looking into a report that the Controller on the ESP32 seems to generate a Connection Request for the same device Windows 10 device it's already connected with the same BD_ADDR. I hope that's just a bug in the Controller.

mringwal, 2022-11-01:
Could you confirm that 893b4ccd704efdc2bbcc03a20a875687e68c3de8 fixes your issue?

AminoACID123, 2022-11-01:
Thanks for your explanation! [893b4cc](https://github.com/bluekitchen/btstack/commit/893b4ccd704efdc2bbcc03a20a875687e68c3de8) fixes the issue. Actually I found this issue when studying the source code of btstack. It hasn't been reproduced on a real device. I think it might be helpful to report this tiny issue though.

mringwal, 2022-11-01:
Of course. Thanks for reporting again.

closed 2022-11-01 by mringwal