bluekitchen/btstack #439: Include value handle in GATT_EVENT_QUERY_COMPLETE event

jimmo, 2022-11-14:
**Is your feature request related to a problem? Please describe.**

In MicroPython, we provide `gattc_read(conn_handle, value_handle)` and `gattc_write(conn_handle, value_handle, data, mode)` for accessing characteristics.

`gattc_read` is implemented with `gatt_client_read_value_of_characteristic_using_value_handle`. We pass the subsequent `GATT_EVENT_CHARACTERISTIC_VALUE_QUERY_RESULT` event data to the MicroPython "read result" event, and then the `GATT_EVENT_QUERY_COMPLETE` event raises the MicroPython "read complete" event.

Similarly, `gattc_write(..., mode=1)` is implemented with `gatt_client_write_value_of_characteristic`, and the `GATT_EVENT_QUERY_COMPLETE` raises the MicroPython "write complete" event.

The problem is that the `GATT_EVENT_QUERY_COMPLETE` event doesn't include the value handle, and so we have no way to include this in the read/write complete event raised to MicroPython.

**Describe the solution you'd like**
I guess having a `gatt_event_query_complete_get_value_handle()` would be great, but I don't know if that's feasible or sensible.

Even better, hopefully I'm just missing something obvious and there's an easy way to solve this without changes to btstack. Knowing the value handle for the `GATT_EVENT_QUERY_COMPLETE` event seems like it would be a requirement for anyone using client read/write-with-response. I'm curious how other users of btstack handle this.

**Describe alternatives you've considered**
I have investigated ways of tracking the active read/write operation for each connection handle in our bindings. This is a little bit awkward because we otherwise keep very little state in our bindings. I don't know what assumptions I can make about the ordering of responses when there are concurrent requests.

**Additional context**
Docs for the Python API: https://docs.micropython.org/en/latest/library/bluetooth.html#bluetooth.BLE.gattc_read
Handler for the `GATT_EVENT_QUERY_COMPLETE` event: https://github.com/micropython/micropython/blob/master/extmod/btstack/modbluetooth_btstack.c#L428
Implementation of `gattc_read`: https://github.com/micropython/micropython/blob/master/extmod/btstack/modbluetooth_btstack.c#L1395

Thanks!

cc @dpgeorge

mringwal, 2022-11-15:
Hi Jim. Let's start with the assumptions/guarantees: with the current GATT Client (which does not support Extended ATT yet), there can be only one Read / Write ATT operation per HCI Connection. 

If you already have some state per HCI connection, you could add the attribute value handle there safely. Would this help?

I guess the attribute value handle wasn't required by other users much as I would expect some state machines in there which execute a sequence of steps, hence, it always knows what Characteristic currently has been written/read based on its internal state (in other words, I expect dispatching based on internal state instead of attribute value handle / operation type).

The GATT_EVENT_QUERY_COMPLETE is emitted for all GATT Client requests. I guess we could add an attribute_handle to that one, although not all operations have an obvious attribute_handle (e.g discovery primary services...).

jimmo, 2022-11-15:
Thanks @mringwal -- yes that additional context does help a lot. I think it sounds like we can make this work on our side then. We do already have a small amount of state, and with that guarantee can be extended to safely include the value handle.

Understood about the state machine. It's a bit tricky here because the additional layer of abstraction (as well as supporting multiple host stacks with the same Python API), and the (with hindsight, slightly questionable) API design of allowing the Python API to queue writes.

Thanks again. I'll close this now, and re-open if we run into issues.

closed 2022-11-15 by jimmo

mringwal, 2022-11-16:
Great!