bluekitchen/btstack #452: Segmentation Fault when sending hci_le_set_extended_scan_response_data command

AminoACID123, 2023-02-01:
**Describe the bug**
In function `hci_run_general_gap_le` in `src/hci.c`, a segmentation fault may happen when sending `hci_le_set_extended_scan_response_data` command
```
  if ((advertising_set->tasks & LE_ADVERTISEMENT_TASKS_SET_SCAN_DATA) != 0) {
      uint16_t pos = advertising_set->scan_data_pos;
      uint8_t  operation = hci_le_extended_advertising_operation_for_chunk(pos, advertising_set->scan_data_len);
      uint16_t data_to_upload = btstack_min(advertising_set->scan_data_len - pos, LE_EXTENDED_ADVERTISING_MAX_CHUNK_LEN);
      if ((operation & 0x02) != 0){
          advertising_set->scan_data_pos = 0;
          advertising_set->tasks &= ~LE_ADVERTISEMENT_TASKS_SET_SCAN_DATA;
      } else {
          advertising_set->scan_data_pos += data_to_upload;
      }
      hci_stack->le_advertising_set_in_current_command = advertising_set->advertising_handle;
      hci_send_cmd(&hci_le_set_extended_scan_response_data, advertising_set->advertising_handle, operation, 0x03, 0x01, data_to_upload, &advertising_set->scan_data[pos]);
      return true;
  }
```
Given that `hci_le_set_extended_scan_response_data` has the format `111JV`, `hci_send_cmd` function should accept 6 arguments in total, but 7 are actually given.

**Environment:**
 - Current BTstack branch: master

mringwal, 2023-02-01:
Thanks for reporting. I did stumble upon that just recently and I have pushed a fix on the develop branch. Please let me know if that works for you.

AminoACID123, 2023-02-01:
That works for me, thanks.

closed 2023-02-01 by AminoACID123