bluekitchen/btstack #455: Problems linking in platform io

Chinchila01, 2023-02-07:
**Describe the bug**

I installed the btstack library to the platformio location but I get a linking error `undefined reference to `btstack_init'`

**To Reproduce**

Steps to reproduce the behavior:
1. Create a platform io project with ESP-IDF
2. Install the btstack library to platform-io's esp-idf platform location
3. Copy the spp_counter source files to the src directory
4. Build the project

**Expected behavior**

The build succeeds.

**HCI Packet Logs**
N/A

**Environment:  (please complete the following information):**
 - Current BTstack branch: master
 - Bluetooth Controller: ESP32
 - Remote device: None

**Additional context**
I have this build log, am I missing any link flags or something? I checked in the example's folder but didn't seem to find any:

`CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50000.0 (5.0.0)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 11.2.0+2022r1
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
xtensa-esp32-elf-gcc -o .pio\build\esp32dev\firmware.elf -T memory.ld -u esp_app_desc -u pthread_include_pthread_impl -u pthread_include_pthread_cond_impl -u pthread_include_pthread_local_storage_impl -u pthread_include_pthread_rwlock_impl -u ld_include_highint_hdl -u start_app -u start_app_other_cores -T sections.ld -u __ubsan_include -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -u __assert_func -u esp_dport_access_reg_read -T esp32.peripherals.ld -u app_main -u newlib_include_heap_impl -u newlib_include_syscalls_impl -u newlib_include_pthread_impl -u newlib_include_assert_impl -u __cxa_guard_dummy -u __cxx_fatal_exception -u include_esp_phy_override -u vfs_include_syscalls_impl -Wl,--Map=C:/Users/gmang/Documents/PlatformIO/Projects/CarBoomboxESPIDF/.pio/build/esp32dev/CarBoomboxESPIDF.map -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--gc-sections -Wl,--undefined=FreeRTOS_openocd_params -Wl,--undefined=uxTopUsedPriority -Wl,--warn-common -Wl,--wrap=_Unwind_Backtrace -Wl,--wrap=_Unwind_DeleteException -Wl,--wrap=_Unwind_FindEnclosingFunction -Wl,--wrap=_Unwind_Find_FDE -Wl,--wrap=_Unwind_ForcedUnwind -Wl,--wrap=_Unwind_GetCFA -Wl,--wrap=_Unwind_GetDataRelBase -Wl,--wrap=_Unwind_GetGR -Wl,--wrap=_Unwind_GetIP -Wl,--wrap=_Unwind_GetIPInfo -Wl,--wrap=_Unwind_GetLanguageSpecificData -Wl,--wrap=_Unwind_GetRegionStart -Wl,--wrap=_Unwind_GetTextRelBase -Wl,--wrap=_Unwind_RaiseException -Wl,--wrap=_Unwind_Resume -Wl,--wrap=_Unwind_Resume_or_Rethrow -Wl,--wrap=_Unwind_SetEnableExceptionFdeSorting -Wl,--wrap=_Unwind_SetGR -Wl,--wrap=_Unwind_SetIP -Wl,--wrap=__cxa_call_unexpected -Wl,--wrap=__deregister_frame_info -Wl,--wrap=__deregister_frame_info_bases -Wl,--wrap=__gxx_personality_v0 -Wl,--wrap=__register_frame -Wl,--wrap=__register_frame_info -Wl,--wrap=__register_frame_info_bases -Wl,--wrap=__register_frame_info_table -Wl,--wrap=__register_frame_info_table_bases -Wl,--wrap=__register_frame_table -Wl,--wrap=longjmp -Wno-frame-address -fno-lto -fno-rtti -mlongcalls -Og -g2 -ggdb2 .pio\build\esp32dev\src\main.o .pio\build\esp32dev\src\rfcomm_client.o -L.pio\build\esp32dev -LC:\Users\gmang\.platformio\packages\framework-espidf\components\esp_wifi\lib\esp32 -LC:\Users\gmang\.platformio\packages\framework-espidf\components\xtensa\esp32 -L.pio\build\esp32dev\esp-idf\esp_system\ld -LC:\Users\gmang\.platformio\packages\framework-espidf\components\esp_rom\esp32\ld -LC:\Users\gmang\.platformio\packages\framework-espidf\components\soc\esp32\ld -LC:\Users\gmang\.platformio\packages\framework-espidf\components\esp_phy\lib\esp32 -LC:\Users\gmang\.platformio\packages\framework-espidf\components\esp_wifi\lib\esp32 -Wl,--start-group .pio\build\esp32dev\esp-idf\xtensa\libxtensa.a .pio\build\esp32dev\esp-idf\app_trace\libapp_trace.a .pio\build\esp32dev\esp-idf\console\libconsole.a .pio\build\esp32dev\esp-idf\unity\libunity.a .pio\build\esp32dev\esp-idf\btstack\libbtstack.a .pio\build\esp32dev\esp-idf\cmock\libcmock.a .pio\build\esp32dev\esp-idf\protocomm\libprotocomm.a .pio\build\esp32dev\esp-idf\espcoredump\libespcoredump.a .pio\build\esp32dev\esp-idf\esp_local_ctrl\libesp_local_ctrl.a .pio\build\esp32dev\esp-idf\wear_levelling\libwear_levelling.a .pio\build\esp32dev\esp-idf\sdmmc\libsdmmc.a .pio\build\esp32dev\esp-idf\json\libjson.a .pio\build\esp32dev\esp-idf\fatfs\libfatfs.a .pio\build\esp32dev\esp-idf\esp_hid\libesp_hid.a .pio\build\esp32dev\esp-idf\protobuf-c\libprotobuf-c.a .pio\build\esp32dev\esp-idf\esp_lcd\libesp_lcd.a .pio\build\esp32dev\esp-idf\mqtt\libmqtt.a .pio\build\esp32dev\esp-idf\perfmon\libperfmon.a .pio\build\esp32dev\esp-idf\spiffs\libspiffs.a .pio\build\esp32dev\esp-idf\wifi_provisioning\libwifi_provisioning.a .pio\build\esp32dev\esp-idf\app_update\libapp_update.a .pio\build\esp32dev\esp-idf\bootloader_support\libbootloader_support.a .pio\build\esp32dev\esp-idf\cxx\libcxx.a .pio\build\esp32dev\esp-idf\driver\libdriver.a .pio\build\esp32dev\esp-idf\efuse\libefuse.a .pio\build\esp32dev\esp-idf\esp-tls\libesp-tls.a .pio\build\esp32dev\esp-idf\esp_adc\libesp_adc.a .pio\build\esp32dev\esp-idf\esp_app_format\libesp_app_format.a .pio\build\esp32dev\esp-idf\esp_common\libesp_common.a .pio\build\esp32dev\esp-idf\esp_eth\libesp_eth.a .pio\build\esp32dev\esp-idf\esp_event\libesp_event.a .pio\build\esp32dev\esp-idf\esp_gdbstub\libesp_gdbstub.a .pio\build\esp32dev\esp-idf\esp_http_client\libesp_http_client.a .pio\build\esp32dev\esp-idf\esp_http_server\libesp_http_server.a .pio\build\esp32dev\esp-idf\esp_https_ota\libesp_https_ota.a .pio\build\esp32dev\esp-idf\esp_hw_support\libesp_hw_support.a .pio\build\esp32dev\esp-idf\esp_netif\libesp_netif.a .pio\build\esp32dev\esp-idf\esp_partition\libesp_partition.a .pio\build\esp32dev\esp-idf\esp_phy\libesp_phy.a .pio\build\esp32dev\esp-idf\esp_pm\libesp_pm.a .pio\build\esp32dev\esp-idf\esp_ringbuf\libesp_ringbuf.a .pio\build\esp32dev\esp-idf\esp_rom\libesp_rom.a .pio\build\esp32dev\esp-idf\esp_system\libesp_system.a .pio\build\esp32dev\esp-idf\esp_timer\libesp_timer.a .pio\build\esp32dev\esp-idf\esp_wifi\libesp_wifi.a .pio\build\esp32dev\esp-idf\freertos\libfreertos.a .pio\build\esp32dev\esp-idf\hal\libhal.a .pio\build\esp32dev\esp-idf\heap\libheap.a .pio\build\esp32dev\esp-idf\http_parser\libhttp_parser.a .pio\build\esp32dev\esp-idf\log\liblog.a .pio\build\esp32dev\esp-idf\lwip\liblwip.a .pio\build\esp32dev\esp-idf\mbedtls\libmbedtls.a .pio\build\esp32dev\esp-idf\newlib\libnewlib.a .pio\build\esp32dev\esp-idf\nvs_flash\libnvs_flash.a .pio\build\esp32dev\esp-idf\pthread\libpthread.a .pio\build\esp32dev\esp-idf\soc\libsoc.a .pio\build\esp32dev\esp-idf\spi_flash\libspi_flash.a .pio\build\esp32dev\esp-idf\tcp_transport\libtcp_transport.a .pio\build\esp32dev\esp-idf\ulp\libulp.a .pio\build\esp32dev\esp-idf\vfs\libvfs.a .pio\build\esp32dev\esp-idf\wpa_supplicant\libwpa_supplicant.a .pio\build\esp32dev\esp-idf\mbedtls\mbedtls\library\libmbedcrypto.a .pio\build\esp32dev\esp-idf\mbedtls\mbedtls\library\libmbedtls.a .pio\build\esp32dev\esp-idf\mbedtls\mbedtls\library\libmbedx509.a -lgcov -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lxt_hal -lc -lm -lstdc++ -lgcc -lphy -lrtc -Wl,--end-group
c:/users/gmang/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.o:(.literal.app_main+0x0): undefined reference to `btstack_init'
c:/users/gmang/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.o: in function `app_main':
C:\Users\gmang\Documents\PlatformIO\Projects\CarBoomboxESPIDF/src/main.c:56: undefined reference to `btstack_init'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1`

mringwal, 2023-02-08:
Hi. I have no experience with Platform IO. I suspect Bluetooth is not enabled in sdkconfig.

Could you switch to the current BTstack develop branch and try your steps again? It should now provide an error message with instructions on how to update sdkconfig.

Chinchila01, 2023-02-08:
Will give it a try and send feedback!

Chinchila01, 2023-02-08:
When I compiled I got the new errors. Ran menuconfig, configured it correctly and its now working!

Thanks!

mringwal, 2023-02-08:
Thanks for testing! Glad you got it working and nice to know that it can be use with PlatformIO.

closed 2023-02-08 by mringwal