bluekitchen/btstack #456: Samsung Galaxy S7, fail to connect BLE HID keyboard on Pico W

matsobdev, 2023-02-14:
Android 8.0.0 blinks message with password to type on Pico W, that sends in serial its own password "Display Passkey: xxxxxxxx". Both numbers are different. Then serial sends "Disconnected". Example "hog_keyboard_demo.c"

mringwal, 2023-02-15:
So, both both sides hows a passkey (which is of course different)? That cannot work.
Could you:
- enable HCI logging by setting WANT_HCI_DUMP to 1 in picow_bt_example_common.c
- try pairing again and store the output into a file
- use btstack/create_packet_log.py to convert the log file into a .pklg file
- post the pklg file here

Instead of BLE HID (HOG), you can also try Classic HID with the hid_keyboard_demo.c (it's pretty much the same demo, just a different Bluetooth mode).

matsobdev, 2023-02-15:
If I'm doing it right it does not change a thing. It compiles sth different, but just spits out in serial:
`BTstack up and running on 28:CD:C1:00:01:0A.
Display Passkey: 91207
Disconnected
`
And yes, as weird as it sounds both: phone and Pico W are giving password to type in opposite device. Mouse HOG fine. Pico gives password to type in Android.
In the file given by you I'm adding
`#define WANT_HCI_DUMP 1`

mringwal, 2023-02-18:
Hi. I'd like to see the HCI dump. Unless you use a J-Link, don't set ENABLE_SEGGER_RTT to 1.
If you set WANT_HCI_DUMP=1, the code in picow_bt_example_common.c should call hci_dump_init(...).
With that, you should get a timestamped log file that can be converted.

matsobdev, 2023-02-18:
No Segger, no nothing again. To be clear, lon in serial terminal?

matsobdev, 2023-02-19:
Hello! Finally some success. Defining WANT_HCI_DUMP 1 only in 'btstack_config.txt' worked out. So That 'CMakeLista.txt' target_compile_definition() might not work.
[dump.zip](https://github.com/bluekitchen/btstack/files/10776410/dump.zip)

matsobdev, 2023-02-19:
Problem kinda solved by setting `sm_set_io_capabilities(SSP_IO_CAPABILITY_NO_INPUT_NO_OUTPUT)`. I'm new to that kind of stuff, but is it lower level of security when bonded?

mringwal, 2023-02-20:
Glad you got the HCI log. 

In the log, your phone is the Initiator and sends a pairing request with "Display / Keyboard" IO Capabilities. hog_keyboard_demo, as Responder, sends a pairing response with "Display Only". If you check in the table you have posted, the Responder (hog_keyboard_demo) should show a Passkey (which it does) and this Passkey should get entered on the Initiator (aka, your phone). This looks like a bug in Android 8. Could you try a newer Android phone or update it?

One way around this, which you've found, is to use "No Input/No Output", which results in "Jus Works" pairing.

In general, if there is no attacker present during pairing, all pairing variants are safe - both sides share a secret AES128 key that no-one else has access to.

If an attacker is present, she could set up a Man in the Middle Attack if "Just Works" is used. With LE Legacy Pairing, it's sufficient to sniff a pairing and then quickly try all possible passkey combinations to get the key.

 %"PRIu32" is the only portable way to print a 32 bit number. It requires #include <inttypes.h>. What error did you get when compiling it?

matsobdev, 2023-02-20:
Thank you for the introduction to the Bluetooth. A only have again Android 8.0.0 but TV decoder. I will post when I will be able to check it, then close the issue. Below compile error:
```
[ 14%] Building C object CMakeFiles/hog_kbd.dir/nordic_spp_le_counter.c.obj
G:\nordic\nordic_spp_le_counter.c: In function 'packet_handler':
G:\nordic\nordic_spp_le_counter.c:158:54: error: expected ')' before 'PRIu32'
  158 |             printf("Confirming numeric comparison: %"PRIu32"\n", sm_event_numeric_comparison_request_get_passkey(packet));
      |                                                      ^~~~~~
      |                                                      )
```
But again, this is funny. It is from SPP like Nordic, that a 'mixed' with HOG mouse to get autenthication and pairing/bonding. It displayed shorter than 6 digits password once, but as I remember trying to debug `hog_keyboard_demo` the same happend when applying `WANT_HCI_DUMP 1`. Bare `hog_mouse_demo.c` and other compiles without an issue. The same toolchain, nothing changed, the same `CMakeLists.txt` only names to the files different.

closed 2023-03-10 by matsobdev

matsobdev, 2024-03-23:
Trying before different SM setups and stuff and keeping in mind, that `hog_mouse_demo.c` was working in default state, so just changing appearance from keyboard to mouse (`0xC1` -> `0xC2`) in `hog_keyboard_demo` was a solution to allow usage of PIN when pairing.

mringwal, 2024-03-24:
Good catch. The HOG device demos use these appearance values already (e.g. hog mouse uses 0xc2)