bluekitchen/btstack #458: Classic HID Device on Pico W is appending an extra 0x00 to the end of the descriptor

sanjay900, 2023-02-16:
I am currently using the `hid_keyboard_demo` on a pi pico w, and have connected to a linux pc, where i noticed i would always see a `unknown main item tag 0x0` error in the terminal. After dumping the hid descriptor from linux, i found that an extra 0x00 byte was sitting at the end of the descriptor.

For example, the keyboard descriptor from this example shows up in linux as
```
Device Path: /dev/hidraw7
Raw Name: SantrollerBT
Raw Phys: 00:15:83:ed:65:e8
Raw Info:
	bustype: 5 (Bluetooth)
	vendor: 0x1209
	product: 0x2885
Report Descriptor Size: 68
Report Descriptor:
05 01 09 06 A1 01 85 01 05 07 75 01 95 08 05 07
19 E0 29 E7 15 00 25 01 81 02 75 01 95 08 81 03
95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03
91 03 95 06 75 08 15 00 25 FF 05 07 19 00 29 FF
81 00 C0 00
```

sanjay900, 2023-02-17:
nevermind i think this might have been my fault

closed 2023-02-17 by sanjay900

mringwal, 2023-02-20:
What do you think was your mistake? 

I double checked and Wireshark shows a descriptor with 67 bytes.

[image]

sanjay900, 2023-02-20:
At first I thought the problem that I had made some buffers too small, but
I ended up switching to BLE instead of classic which ended up solving my
issue.

I also noticed that Wireshark was getting the packet correctly without the
extra byte, so I figured this might just be something odd happening in
linux. I also noticed that Linux was just ignoring the extra byte anyways,
and people have reported Linux doing this with other Bluetooth devices too.

mringwal, 2023-02-20:
Thanks for the explanation.

matsobdev, 2023-03-14:
There is double:
```
0x05, 0x07,                    //   Usage Page (Key codes)
```
in HID example by the way in `// Modifier byte (input)`