bluekitchen/btstack #459: ESP32 cannot pair bluetooth with any device

Chinchila01, 2023-02-17:
**Describe the bug**

I'm using code from a tutorial to try and get the ESP32 to pair with any other of my devices but I can't.
I get this error:

[image]

**To Reproduce**

Steps to reproduce the behavior:
1. Run this code:
```#include "btstack.h"
#include <stdio.h>
 
static void packet_handler (uint8_t packet_type, uint16_t channel, uint8_t *packet, uint16_t size);
 
static btstack_packet_callback_registration_t hci_event_callback_registration;
 
static void packet_handler (uint8_t packet_type, uint16_t channel, uint8_t *packet, uint16_t size){
 
   uint16_t rfcomm_channel_id;
 
   if (packet_type == HCI_EVENT_PACKET
       && hci_event_packet_get_type(packet) == RFCOMM_EVENT_INCOMING_CONNECTION){
 
       rfcomm_channel_id = rfcomm_event_incoming_connection_get_rfcomm_cid(packet);
       rfcomm_accept_connection(rfcomm_channel_id);
 
     }else if (packet_type == RFCOMM_DATA_PACKET){
 
        printf("Received data: '");
 
        for (int i=0;i<size;i++){
             putchar(packet[i]);
         }
 
          printf("\n----------------\n");
    }
}
 
int btstack_main(int argc, const char * argv[]){
 
    hci_event_callback_registration.callback = &packet_handler;
    hci_add_event_handler(&hci_event_callback_registration);
 
    l2cap_init();
 
    rfcomm_init();
 
    rfcomm_register_service(packet_handler, 1, 0xffff); 
 
    sdp_init();
 
    gap_discoverable_control(1);
 
    hci_power_control(HCI_POWER_ON);
 
    return 0;
}```
2. Try to pair from PC or any other device

**Expected behavior**

Device should pair.

**HCI Packet Logs**
Here is the full log:
[btstack-full-log.txt](https://github.com/bluekitchen/btstack/files/10766365/btstack-full-log.txt)

**Environment:  (please complete the following information):**
 - Current BTstack branch: develop
 - Bluetooth Controller: ESP32
 - Remote device: PC with Windows 10

**Additional context**
None

mringwal, 2023-02-18:
Hi. In the log, there's no communication with the PC. When you try to connect from the PC, you should see some traffic.
From your pasted code, SDP is initialized but there are no SDP records registered. Most devices will query SDP and look for service that they support. If you don't offer any, that won't work. Please try the spp_counter demo (it sets up an RFCOMM server and registers an SDP record for it).

closed 2023-02-18 by mringwal

Chinchila01, 2023-02-20:
Hey,
I tried with the spp_counter demo and I'm getting the same behavior:

here are the logs, no packets are received when trying to connect.

```
[00:00:00.545] LOG -- hci.c.5064: hci_power_control: 1, current mode 0
[00:00:00.555] LOG -- main.c.213: transport_init
[00:00:00.555] LOG -- main.c.229: transport_open
␛[0;32mI (565) BTDM_INIT: BT controller compile version [8020d24]␛[0m
␛[0;32mI (565) system_api: Base MAC address is not set␛[0m
␛[0;32mI (575) system_api: read default base MAC address from EFUSE␛[0m
␛[0;32mI (585) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07␛[0m
[00:00:00.915] LOG -- hci.c.7519: BTSTACK_EVENT_STATE 1
[00:00:00.915] EVT <= 60 01 01
[00:00:00.915] CMD => 03 0C 00
[00:00:00.925] EVT <= 6E 00 
[00:00:00.925] EVT <= 0E 04 05 03 0C 00
[00:00:00.925] CMD => 01 10 00
[00:00:00.925] EVT <= 0E 0C 05 01 10 00 08 0E 03 08 60 00 0E 03
[00:00:00.935] LOG -- hci.c.2780: Manufacturer: 0x0060
[00:00:00.935] EVT <= 6E 00
[00:00:00.935] CMD => 14 0C 00
[00:00:00.945] EVT <= 0E FC 05 14 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:01.005] LOG -- hci.c.2630: local name: 
[00:00:01.015] EVT <= 6E 00
[00:00:01.015] CMD => 02 10 00
[00:00:01.025] EVT <= 0E 44 05 02 10 00 BF FF FF 03 CC FF EF FF 3F FF FC 1F F2 0F E8 FE 3F F7 8F FF 1C 00 04 00 61 F7 FF FF 7F 38 C0 FF 7F FF FF 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:01.035] LOG -- hci.c.2598: Command 0x00 supported 2/5
[00:00:01.045] LOG -- hci.c.2598: Command 0x01 supported 10/4
[00:00:01.045] LOG -- hci.c.2598: Command 0x02 supported 14/7
[00:00:01.055] LOG -- hci.c.2598: Command 0x03 supported 18/3
[00:00:01.065] LOG -- hci.c.2598: Command 0x04 supported 20/4
[00:00:01.065] LOG -- hci.c.2598: Command 0x05 supported 22/2
[00:00:01.075] LOG -- hci.c.2598: Command 0x06 supported 24/6
[00:00:01.075] LOG -- hci.c.2598: Command 0x07 supported 32/1
[00:00:01.085] LOG -- hci.c.2598: Command 0x08 supported 32/3
[00:00:01.085] LOG -- hci.c.2598: Command 0x09 supported 32/6
[00:00:01.095] LOG -- hci.c.2598: Command 0x0a supported 34/0
[00:00:01.095] LOG -- hci.c.2598: Command 0x0b supported 35/1
[00:00:01.105] LOG -- hci.c.2598: Command 0x0c supported 35/3
[00:00:01.105] LOG -- hci.c.2603: Local supported commands summary 00001fff
[00:00:01.115] LOG -- btstack_crypto.c.1121: controller supports ECDH operation: 1
[00:00:01.125] EVT <= 6E 00
[00:00:01.125] CMD => 09 10 00
[00:00:01.135] EVT <= 0E 0A 05 09 10 00 B6 AD C8 D1 08 F0
[00:00:01.135] LOG -- hci.c.2727: Local Address, Status: 0x00: Addr: F0:08:D1:C8:AD:B6
[00:00:01.145] EVT <= 6E 00
[00:00:01.145] CMD => 05 10 00
[00:00:01.155] EVT <= 0E 0B 05 05 10 00 FD 03 FF 09 00 04 00 
[00:00:01.155] LOG -- hci.c.2645: hci_read_buffer_size: ACL size module 1021 -> used 1021, count 9 / SCO size 255, count 4      
[00:00:01.165] EVT <= 6E 00 
[00:00:01.165] CMD => 03 10 00
[00:00:01.175] EVT <= 0E 0C 05 03 10 00 BF EE CD FE DB FF 7B 87
[00:00:01.175] LOG -- hci.c.2761: Packet types cc18, eSCO 1
[00:00:01.185] LOG -- hci.c.2764: BR/EDR support 1, LE support 1
[00:00:01.185] EVT <= 6E 00
[00:00:01.185] CMD => 33 0C 07 00 04 3C 14 00 0A 00
[00:00:01.195] EVT <= 0E 04 05 33 0C 00
[00:00:01.195] EVT <= 6E 00
[00:00:01.195] CMD => 31 0C 01 03
[00:00:01.205] EVT <= 0E 04 05 31 0C 00
[00:00:01.205] EVT <= 6E 00
[00:00:01.205] CMD => 01 0C 08 FF FF FF FF FF FF FF 3F 
[00:00:01.215] EVT <= 0E 04 05 01 0C 00
[00:00:01.215] EVT <= 6E 00
[00:00:01.225] CMD => 63 0C 08 00 00 00 02 00 00 00 00
[00:00:01.225] EVT <= 0E 04 05 63 0C 00 
[... 103 lines cut ...]
[00:00:01.745] EVT <= 6E 00 
[00:00:01.745] LOG -- btstack_crypto.c.956: more ecc random
[00:00:01.745] CMD => 18 20 00
[00:00:01.755] EVT <= 0E 0C 05 18 20 00 1E 76 D9 76 88 98 6C 49 
[00:00:01.755] LOG -- btstack_crypto.c.525: set uECC RNG for initial key generation with 64 random bytes
[00:00:01.765] LOG -- btstack_crypto.c.498: sm_generate_f_rng: size 32 - offset 0
[00:00:01.925] EVT <= 6E 00 
[00:00:01.925] LOG -- btstack_crypto.c.488: Elliptic curve: X
[00:00:01.925] LOG -- btstack_util.c.274: 0x51, 0x84, 0xF3, 0xBF, 0xD3, 0xF7, 0xCB, 0xA1, 0xCC, 0xB7, 0xB5, 0xEE, 0x98, 0x2E, 0x17, 0xAF,
[00:00:01.935] LOG -- btstack_util.c.274: 0x2E, 0xDD, 0x19, 0x39, 0xBD, 0x60, 0xE0, 0x0F, 0xB9, 0x85, 0x48, 0xD8, 0x8C, 0xEA, 0x95, 0xD6,
[00:00:01.945] LOG -- btstack_crypto.c.490: Elliptic curve: Y
[00:00:01.945] LOG -- btstack_util.c.274: 0x65, 0x6E, 0x55, 0x27, 0x97, 0x93, 0x4F, 0x1F, 0x1C, 0x83, 0x83, 0x1D, 0x45, 0xB1, 0xA4, 0xBD, 
[00:00:01.965] LOG -- btstack_util.c.274: 0xB8, 0x98, 0x0E, 0x7B, 0xAB, 0xD4, 0x74, 0x56, 0xF2, 0xC1, 0xDE, 0x74, 0x7C, 0xE5, 0x38, 0x20,
[00:00:01.975] CMD => 17 20 20 1E DD 53 31 68 EC 0E DF 65 C9 3D 59 46 D8 EF B5 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[... cut ...]
[00:00:01.995] LOG -- btstack_util.c.314: irk    E72795669B895351CB37F4FA57019CB6
[00:00:01.995] EVT <= 6E 00
[00:00:02.005] LOG -- sm.c.2146: DKG_CALC_DHK started
[00:00:02.005] CMD => 17 20 20 1E DD 53 31 68 EC 0E DF 65 C9 3D 59 46 D8 EF B5 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  
[00:00:02.025] EVT <= 0E 14 05 17 20 00 BD F6 A9 51 81 D2 E7 8F 38 C8 24 54 D0 F3 78 7C
[00:00:02.025] LOG -- btstack_util.c.314: dhk    7C78F3D05424C8388FE7D28151A9F6BD
[00:00:02.035] EVT <= 6E 00
```

How should I go about finding out what's wrong with my setup?

mringwal, 2023-02-21:
As there's no connection, I guess there are two options:
a) there's a second device with the same Bluetooth MAC address that your Windows PC connects to (so we don't see it in this log)
b) your Windows PC is confused and rejects the connection without even trying. I would suggest to use a different PC / Android phone to check if it's a problem with this particular PC. As far as I know, Windows stores device information in the Registry. You can try to find and delete all related registry entries and try again. And/or reboot.. :)

Chinchila01, 2023-02-21:
I tried pairing with my phone and it doesn't work as well. I'll try another ESP32

mringwal, 2023-02-21:
Ah yes... a non-working ESP32 might be another explanation.

Chinchila01, 2023-02-27:
I found the issue:
I had to go back to menuconfig, component config, and change bluetooth controller mode to Bluetooth Dual Mode as it was in BLE only.
That fixed it, thanks for the help!

mringwal, 2023-02-27:
Glad you found it. In the log above, it says
hci.c.2764: BR/EDR support 1, LE support 1
so, I'm not sure if this is a different problem. Well, let's see.