bluekitchen/btstack #460: Links on https://bluekitchen-gmbh.com/btstack/#profiles/ at least for HID are wrong

daveHitchman, 2023-02-22:
**Describe the bug**
Documentation bug, links arent all correct.

**To Reproduce**
 Go to https://bluekitchen-gmbh.com/btstack/#profiles/
Find the HID section
Try any of the links provided.

Example the Hid keyboard demo link is 
https://bluekitchen-gmbh.com/btstack/#profiles/examples/examples/#sec:hidkeyboarddemoExample
appears it would need to be 
https://bluekitchen-gmbh.com/btstack/#examples/examples/#sec:hidkeyboarddemoExample

Similar issues with the others, seems there may have been a slight rearrangement and these links werent fixed

mringwal, 2023-03-03:
Hi there. On the develop branch (resp. its html manual), the links to source files as well as others have been fixed. Please let us know if you find other dead links

closed 2023-03-03 by mringwal

daveHitchman, 2023-03-03:
super :)