bluekitchen/btstack #462: Pairing with "Numeric Comparison" misses the passkey.

felixdoerre, 2023-03-04:
**Describe the bug**

Seemingly since ea9b67963c9 the passkey is not sent with the `SM_EVENT_NUMERIC_COMPARISON_REQUEST`-event, but the memory for the passkey is uninitialized. It seems that refactoring broke the logic and not test case caught it.

**To Reproduce**

Steps to reproduce the behavior:
1. Configure the stack to use secure connections, and io=display_yes_no
2. Initiate a pairing from a remote device that has e.g. display_keyboard
3. Receive the `SM_EVENT_NUMERIC_COMPARISON_REQUEST`
4. Extract the passkey from that event with `sm_event_numeric_comparison_request_get_passkey`
5. Notice that it not only does not match the key from the remote device, but also is out of range (e.g. `-731160572` in one run)

**Expected behavior**

In step "5" I would expect the passkey to be the same as shown on the remote device.

**Environment:  (please complete the following information):**
 - Micropython with btstack at: 1635e36d0
 - Bluetooth Controller CYW43 on Pico W
 - Remote device: Android

**Additional context**
I believe the refactoring broke the `NUMERIC_COMPARISON` case and `sm_trigger_user_response_passkey` should be extended to also be usable for `SM_EVENT_NUMERIC_COMPARISON_REQUEST`.

mringwal, 2023-03-06:
Hi Felix. Thanks for the issue report and the bi-sect. You're totally right, when extracting the helper functions, the value for the Numeric Comparison event got lost. It's fixed on the develop branch. Could you confirm the fix?

felixdoerre, 2023-03-06:
I've cherry-picked the fix locally on v1.5.4, tested a numeric comparison pairing and it resolves the issue. Thanks.

commit 74b4d42a (2023-03-06): sm: fix value in SM_EVENT_NUMERIC_COMPARISON_REQUEST

Regression introduced in ea9b679, fixes #462
closed 2023-03-06 by felixdoerre