bluekitchen/btstack #465: Best Practice for BLE GATT UART

savejeff, 2023-04-03:
since recently the Raspberry Pi Pico W received Bluetooth support and using the arduino-pico Arduino RP2040 core uses btstack (maybe even the Pico SDK uses it, I'm not sure)

I started porting a working project from ESP32 to the RP2040. Here i use the quasi-standart UART UUID (introduced by nordic)

```
#define SERVICE_UUID_UART				"6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_UART_RX		"6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_UART_TX		"6E400003-B5A3-F393-E0A9-E50E24DCCA9E"
```
with a RX characteristic with only WRITE and a TX characteristic with only NOTIFY

i followed the Arduino Example  "LE Peripheral Example" but it seems the current Wrapper class BTstackManager does not support notify yet.
I've also gone through some of the example here but did not find an example that matched. 

So my question: what would be the recommended way to implement the UART over BLE GATT? 
As the UART Service is more widely used, would it be possible to add an example for it ?

sonnny, 2023-04-03:
I have 2 example of pico w uart using two way comms demo in youtube. Links below, I use and android app to connect to pico w, source in the description. The demo shows ble and classic bluetooth.

[ble demo](https://www.youtube.com/watch?v=teNcl7KLZeg)

[bluetooth classic demo](https://www.youtube.com/watch?v=ko_IgcIizaU)

savejeff, 2023-04-04:
okay thx. ill look into the source for the nordic spp streamer example

i only looked into the examples marked with le gatt. 
currently this is only a weekend project and i wanted to ask if i overlooked something before sinking in too much time for something that already has an example code.

i think it would be helpful to others to have a generic gatt_uart example, as its a widely used convention. there are app for android/ios from adafruit that use that convention.

mringwal, 2023-06-05:
So, what's the verdict here? we have examples for u-blox and nordic's generic spp. unfortunately, there's no standard for SPP over GATT either. Nordic's allow to use their nRF Toolkit on smartphone, so that's an easy start.

savejeff, 2023-06-05:
A yes sorry I was to busy for this project recently. I think I roughly now what I should do. 
I'll close the issuey but I'm still recommending for a basic GATT UART example with an easy to find name. The UUID are commonly used and it would be grear to have something that works unchanged with available ble UART apps from the app store

closed 2023-06-05 by savejeff

mringwal, 2023-06-05:
@savejeff what are 'available ble uart apps from the app store" for you? iOS or Google? nRF Toolbox is on both OSs.

savejeff, 2023-06-05:
For the GATT UART there are multiple apple on both app stores. 
For example the adfruit App https://play.google.com/store/apps/details?id=com.adafruit.bluefruit.le.connect

mringwal, 2023-06-05:
Thanks. Looks like Adafruit also uses Nordic's simple UART over GATT version, too. So, it's well supported, but it seems not to be as accessible as it could be.

savejeff, 2023-06-05:
In the Arduino and embadded community it's basically a standard. 
It's also not that hard to implement. You just need the write for the Rx and the notify for the Tx characteristic and you habe a bidirectional byte stream