bluekitchen/btstack #466: BLE Midi connection drops with KORG Microkey2 AIR due to timing issues

shuichitakano, 2023-04-09:
Hello,
I am writing code for BLE Midi (central) on the Raspberry Pi Pico W. While it works with some devices, I am experiencing an issue with the KORG Microkey2 AIR where the connection drops after a short period of time.

Upon capturing packets and analyzing the situation, I confirmed that non-EmptyPDU packets are continuously being sent from the master side.
[image]

However, when I define #define WANT_HCI_DUMP 1 and build the code, it works without any issues, and I can see that the EmptyPDU packets are being sent.
It appears to be a timing issue.

By buffering the printf output in memory instead of printing it immediately, and displaying it at the end, I was able to reproduce the issue while still capturing logs. It seems that the order of some packet generation is different.
[image]

Do you have any hints or insights on what might be happening?
[btlog.zip](https://github.com/bluekitchen/btstack/files/11185133/btlog.zip)

mringwal, 2023-04-09:
Hi. Glad you were able to also get air traces. 
In both cases, the connections breaks shortly after a connection parameter update initiated by the peripheral.
In one case, you get a connection timeout, in the other a 'inacceptable connection parameters' - although the params are provided by the peripheral and BTstack uses the same in both cases.

anyway,. it looks like an issue between the link layers of both Controllers. I would suggest to try to block the connection parameter update. The peripheral requests 7.5-8.75ms and zero supervision timeout. The 7.5 ms is probably for minimal latency. You can set the range that's accepted by the stack with gap_get_connection_parameter_range. Here, I'd use a higher min conn latency, which will cause the con param update to get rejected and just use 7.5 and con latency of 4 initially in gap_set_connection_parameters

Please let me know if that helps

shuichitakano, 2023-04-15:
Thank you for your suggestion.
By using the gap_set_connection_parameter_range function and setting le_conn_latency_min to 1 or greater, the connection no longer drops!

However, I am still concerned about the latency. 
It seems that I can balance both connection stability and low latency by rejecting the Connection Parameter Update Request from the slave and then setting a slightly more conservative latency using gap_update_connection_parameters.
Ideally, I would like to know the timing of the rejection of the Connection Parameter Update Request and update the settings accordingly, but it seems that there are no events being raised for this. Is there any way to know the timing?

For now, I've set it to update on the first notification, which seems to work fine in practice.
Thank you for your help!

mringwal, 2023-04-15:
Thanks for the confirmation. We've seen the 'dropped connection' in other combinations as well.
For your BLE MIDI use case, I guess it would make sense to initiate connection with ideal connection parameters (7.5 ms, slave latency of 2). As far I remember, the slave latency allows the peripheral to skip connection events. This should not have a negative impact for latency for its notification (the KORG Microkey2 AIR being an input device of sorts), as its controller will use the next connection interval when it has data to send independent from the slave latency setting.

As for your question: yes, there's currently no event for the result of the con param update request. If you can come up with a clear case where it's needed, we will try to add one. ;)

closed 2023-04-15 by mringwal