bluekitchen/btstack #467: Decreasing hog host latency

sanjay900, 2023-04-10:
Hello, I have two Pico was, one configured as a hog host and one as a hog device, using code similar to the hog hid keyboard examples, only with descriptors swapped out for hid gamepad ones. 

When I use the device with a PC, I can measure a effect poll rates of around 800hz, while when using it with the hog host it ends up closer to 7.5ms or so, but only after dropping the min connection interval to 7.5ms. 

Is there some way to increase the poll rate of the hog host?

mringwal, 2023-04-10:
Hi. Please post on the community mailing list. We'd like to use GitHub issues for issues with the stack itself. By posting on the mailing list, chances are higher that other people jump in or learn from the discussion.

sanjay900, 2023-04-10:
Sorry will do that now did not realise that was a thing

closed 2023-04-10 by sanjay900

mringwal, 2023-04-11:
Thanks for moving over to the mailing list.

sanjay900, 2023-04-12:
Yeah, same issue