bluekitchen/btstack #474: Simpler standalone BT example code for the Raspberry Pico

olimaticer, 2023-05-09:
Hi

I was able to compile separately “spp_counter.c” as BT classic in conjunction with the “standalone” directory under Windows. 
In my eyes, whoever wrote the BT example code for the Raspberry Pico is a true artist of CmakeLists.txt files. Respect! 

**I think I am not the only one** who need a simpler and shorter example to figure out what is going on.
It should be a standalone outside the lib examples one but not so complex like the one within the standalone folder.

A BT classic SPP example does the trick.

An enhanced blink.c which sends an on/off message to the asynchronous BT task in the second c file and get a message from the asynchronous BT task to switch on/off an another LED.

Some code comments about the message the transfer between the two parts also within CmakeLists.txt are welcome.

Who can help? 
Matthias you?

Bests
Gregor

peterharperuk, 2023-05-09:
There are a couple of standalone examples (simple gatt stuff) in here https://github.com/raspberrypi/pico-examples/tree/master/pico_w/bt/standalone

mringwal, 2023-05-10:
Hi Gregor. I've played with the a2dp + cosmic unicorn demo https://github.com/Gadgetoid/galactic-bluetooth-audio
It contains a copy of our a2dp_sink_demo and the necessary files to use the cosmic unicorn device.

The BTstack examples in the example/ folder are supposed to be called via btstack_main (and are fully portable) after a port-specific setup (e.g. for the Pico W).

sonnny, 2023-07-17:
Hi Gregor,

     Not sure if I know what you want, but I have a couple of example, link below for a simpler main.c

[ble template](https://github.com/sonnny/picow_ble_template)

[with youtube demo](https://github.com/sonnny/picow_bluetooth_androidapp_toycar)