bluekitchen/btstack #475: H4 SPI for Zephyr

KlemenDEV, 2023-05-10:
Is it possible to use this stack using H4 HCI protocol via SPI for connection to a Zephyr-based HCI_SPI controller?

I have checked hci_transport.h and based on hci_transport_config_type_t only USB and UART are supported. Is it possible to use this project for SPI-based communication and if yes, where should I start looking for some guidance?

mringwal, 2023-05-10:
Good question. HCI over SPI isn't standardized, but if you use SPI Master, there are not too many options on how you can send HCI packets over it.

As far as I know, Zephyr implements the H4 SPI from ST Micro's BlueNRG Controller.

In any case, we currently don't have an implementation for it, but we have an H4 SPI for EM Microelectronics Controllers (src/hci_transport_em9304_spi.c) including an relative generic  SPI driver (src/btstack_spi_em9304.c) and we also have "H4 Zephyr" on our "nice to have" list.

Which Controller do you want to use? I guess it's possible to use the Controller with Zephry's H4 UART implementation.

What do you use as a host? Asides from an embedded linux system, most OS usually lack support for SPI + GPIO.

Please have a look at the mentioned sources. The closest port is probably the port/stm32-l073rz-nucleo-em9304.

I've been wondering if it makes sense to add support for GPIOs and then use the MultiBus project for a desktop implementation:  https://github.com/mringwal/multibus

KlemenDEV, 2023-05-10:
Thank you for the quick response and the details :)

I would like to use it with the STM32 host (with a custom RTOS on it). The controller is one of the nRF modules running Zephyr in HCI_SPI mode. The controller can indeed run in both UART and SPI modes.

I prefer to use SPI because it technically allows for faster data rates. But if this complicates things a lot, I guess I could also use UART.

> we also have "H4 Zephyr" on our "nice to have" list.

In case I go with H4 UART (nRF controller to STM32 host), what would be the starting point/reference chipset, in this case, you would recommend?

mringwal, 2023-05-10:
> I prefer to use SPI because it technically allows for faster data rates. But if this complicates things a lot, I guess I could also use UART.

Well, although it depends on your application, I haven't seen faster than 1 mbps transfers over LE, so the 1 mbps UART of the nRF5x should not be a real issue.

For H4 UART, you could look at the stm32-f4 port with CC2564 (dual mode), and after getting it to work, disable support for Classic. Please also check our manual as well as the blog posts, esp. the one on the F4 port.

KlemenDEV, 2023-05-15:
I have sort of successfully ported "stm32-f4 port with CC2564 (dual mode)" to our use case with L4 STM32 and Zephyr via UART H4.

However, now I am trying to run the gap_inqury.c example and have hit a roadblock. No devices are found and the sample stays at "Starting inquiry scan..".

You have instructed to disable Classic

> disable support for Classic. Please also check our manual as well as the blog posts, esp. the one on the F4 port.

mringwal, 2023-05-15:
Hi. The nRF5x Controllers are LE-only, that's why I suggested to disable Classic. gap_inquiry demo does a Classic Bluetooth device discovery. Please try gap_le_advertisments.c instead or gatt_counter which implements a simple LE Peripheral.

KlemenDEV, 2023-05-15:
It works now, thank you for the assistance :)

Unless you want to keep this open in case once SPI is supported one day, from my side this issue can be closed.

KlemenDEV, 2023-05-16:
So I think I still have some problems.

While gap_le_advertisments.c worked perfectly (finds all other nearby BLE devices), gatt_counter.c does not work for me. 

I am building it with gatt_counter.c and gatt_counter.h generated using compile_gatt.py. It builds fine and logs show no errors, but the BLE scanner app on my phone does not find the peripheral or its name.

Here is m gatt_counter.h: https://pastebin.com/VVTq1hqY

Here is the log with ERR,INFO, and DEBUG enabled: https://pastebin.com/E61c8DRS

I have also tried nordic and ublox counter examples, but none work. Seems I am missing something to be able to use the Zephyr controller in peripheral mode.

BTStack config header file is the same as in my comment above.

Based on the log, it also seems beat is only updated once and then no longer updates.

mringwal, 2023-05-16:
Zephyr nRF Controller does not have a unique public BD ADDR and e.g. iOS ignores advertisements from devices with BD ADDR 00:00:00:00:00:00. Either enable (non-)resolvable private addresses or enable Zephyr Chipset support and use the unique address stored in the Controller as shown by port/posix-h4-zephyr/main.c.

KlemenDEV, 2023-05-19:
Right now I am exploring the library itself using UART, but I will definitely look into your SPI implementation once this is done, thanks for your work.

Zephyr uses BlueNRG-inspired HCI via SPI so technically it should work to some degree.

KlemenDEV, 2023-05-24:
I have ran into some issues that are probably related to the Zephyr side, so I may not be able to test SPI straight away: https://github.com/zephyrproject-rtos/zephyr/issues/58236

mringwal, 2023-05-24:
Well, zephyr could need them. I'm curious, too, but can check there as well. :) At Zephyr repo.