bluekitchen/btstack #476: Build issue building pico-examples with develop version of btstack

peterharperuk, 2023-05-18:
If I build pico-examples I get build errors, e.g.

/home/peterh/source/pico/pico-sdk2/lib/btstack/src/hci.c: In function 'hci_run_general_pending_commands':
/home/peterh/source/pico/pico-sdk2/lib/btstack/src/hci.c:7131:39: error: 'hci_stack_t' has no member named 'le_minimum_ce_length'
 7131 |                              hci_stack->le_minimum_ce_length, hci_stack->le_maximum_ce_length);
      |                                       ^~
/home/peterh/source/pico/pico-sdk2/lib/btstack/src/hci.c:7131:72: error: 'hci_stack_t' has no member named 'le_maximum_ce_length'
 7131 |                              hci_stack->le_minimum_ce_length, hci_stack->le_maximum_ce_length);
      |                                                                        ^~
/home/peterh/source/pico/pico-sdk2/lib/btstack/src/hci.c:7137:39: error: 'hci_stack_t' has no member named 'le_minimum_ce_length'
 7137 |                              hci_stack->le_minimum_ce_length, hci_stack->le_maximum_ce_length);
      |                                       ^~
/home/peterh/source/pico/pico-sdk2/lib/btstack/src/hci.c:7137:72: error: 'hci_stack_t' has no member named 'le_maximum_ce_length'
 7137 |                              hci_stack->le_minimum_ce_length, hci_stack->le_maximum_ce_length);

It looks like le_minimum_ce_length and le_maximum_ce_length only exists if ENABLE_LE_CENTRAL is defined, but it's used even without this?

mringwal, 2023-05-18:
Good catch. They are also used by all connection param updates. To fix this, I've moved them into the ENABLE_BLE scope on develop. Please confirm the fix.

mringwal, 2023-05-19:
Excellent. Thanks.

closed 2023-05-19 by mringwal