bluekitchen/btstack #477: Failing reestablish encryption (sm.c.4753: Unexpected PDU 11 in state 25) using whitelist

amorniroli, 2023-05-19:
**Describe the bug**

Testing bonding & encryption reestablish using whitelist and resolving list features.

Bonding succeed without problems; next connections, when trying to reestablish security (sm_request_pairing) unexpected error is reported by sm.c module (sm.c.4753: Unexpected PDU 11 in state 25). Mobile phone (nrf connect test app) is then deleting bonding information.

**To Reproduce**

Steps to reproduce the behavior:
1) [sources.zip](https://github.com/bluekitchen/btstack/files/11518876/sources.zip) it's sm_pairing_central.c a bit modified (see the attached one ). See also btstack_config.h attached  (ENABLE_LE_PRIVACY_ADDRESS_RESOLUTION). 

2) first all of, compile using WHITE_USE = 0 (in order to scan without filter and bond to mobile phone);
3) bonding should be succeed. Recompile then using WHITE_USE = 1. Also add 

`gap_whitelist_add (peer_identity_addr_type, peer_identity_addreses);` in file hci.c (I'm lazy, still have to figure out how to add devices to whitelist).

[image]

4) start application. When SM_EVENT_IDENTITY_RESOLVING_SUCCEEDED is triggered, sm_request_pairing function will be called.
5) reestablish encryption will fail due to 

`[00:00:03.095] LOG -- sm.c.4753: Unexpected PDU 11 in state 25`

nrf connect app is also deleting bonding after this

[image]

**Expected behavior**

Encryption should be succeed.

**HCI Packet Logs**
[bonding_log.txt](https://github.com/bluekitchen/btstack/files/11518864/bonding_log.txt) (running app with WHITE_USE  == 0) 
[fail_log.txt](https://github.com/bluekitchen/btstack/files/11518866/fail_log.txt) (running app with WHITE_USE  == 1)

**Environment:  (please complete the following information):**
 - develop branch (0c2c5f79967e853be01f792bf2c33b59fefb0357)
 - ST BlueNRG2
 - samsung a33 android 13 w/ nrf connect app

**Additional context**
Not sure if it's a stack or controller problem.

amorniroli, 2023-05-19:
Tested also with CYBT-343026-EVAL (CYW20706).

Same problem (Unexpected PDU 11 in state 25) but at least mobile phone is not deleting the bonding keys.

Furthremore, SM_EVENT_REENCRYPTION_COMPLETE is not beeing triggered.

Attached the logs with Infineon chipset.

[cyw20706_bonding.txt](https://github.com/bluekitchen/btstack/files/11519200/cyw20706_bonding.txt)
[cyw20706_unexpected_pdu.txt](https://github.com/bluekitchen/btstack/files/11519199/cyw20706_unexpected_pdu.txt)

Alessandro

mringwal, 2023-05-19:
Thanks for the logs. The peripheral is sending an Security Request which we didn't expect in this state, (but it can be safely ignored as we're already taking care of security.

Please try the fix on develop, it should fix the problem for the 20706.

With the BlueNRG2, you get a 'HCI Command Disallow' for the 'HCI LE Start Encryption', which is truly unexpected.
The spec only mentions command disallowed for connections with an associated Connected Isochronous Stream, which we don't have here.

After verifying the fix for the unexpected SM PDU, could you try to ask STM support if they have an idea why we get that response?
Also: there's also a Controller Busy for LE Read Remote Used Features, which is also unexpected. I'll attach the converted packet log in Wireshark format.

[fail_log.pklg.zip](https://github.com/bluekitchen/btstack/files/11519980/fail_log.pklg.zip)

amorniroli, 2023-05-19:
Hi @mringwal ,

Thanks for the feedback. I'll give you some feedback for both Infineon and ST on monday.

Have a nice weekend.

Alessandro

amorniroli, 2023-05-22:
Hi @mringwal ,

I confirm that fix is working for the 20706.

I'll try to ask for ST support and let you know.

Thank you.

Alessandro

amorniroli, 2023-05-22:
> After verifying the fix for the unexpected SM PDU, could you try to ask STM support if they have an idea why we get that response? Also: there's also a Controller Busy for LE Read Remote Used Features, which is also unexpected.

I've opened a case. Meanwhile, trying to add a delay after hci_send_cmd(&hci_le_encrypt, key_flipped, plaintext_flipped); seems to resolve the problem (no more busy error from the controller).

mringwal, 2023-05-22:
Thanks for testing, closing this here. I'll get back to you about the other issues directly.

closed 2023-05-22 by mringwal