bluekitchen/btstack #478: SRC_BLE_GATT_SERVICE_FILES is wrong?

peterharperuk, 2023-06-01:
I don't use this myself but on master btstack/src/ble/gatt-service/Makefile.inc includes the following lines...

    mesh_provisioning_service_server.c \
    mesh_proxy_service_server.c \

These files have been moved from the ble folder to mesh?

mringwal, 2023-06-01:
Hi @peterharperuk - could you double-check? in the current file https://github.com/bluekitchen/btstack/blob/master/src/ble/gatt-service/Makefile.inc I don't see any mesh_* files.

peterharperuk, 2023-06-01:
Groan - you're right. I was looking at the wrong directory. Sorry to waste your time.

closed 2023-06-01 by peterharperuk

mringwal, 2023-06-01:
no worries - it took only a quick look to confirm my suspicion that I had already sorted that.