bluekitchen/btstack #479: Convert GATT service - tool script not working

mklemarczyk, 2023-06-05:
**Describe the bug**

Using the convert_gatt_service.py seems not to work anymore due to changes in the Bluetooth website.

The following url is not valid anymore: https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Services/org.bluetooth.service.reference_time_update.xml
It gets the 404 error.

**To Reproduce**

Steps to reproduce the behavior:
1. Run tool script 'tool/convert_gatt_service.py'

**Expected behavior**

GATT file generated for given service.

**Environment:  (please complete the following information):**
 - Current BTstack branch: master

**Additional context**
The last update of the script seems to date 2020-07-22.

More over there is no given documentation on the format of the *.gatt files.
The fix in form of the full documentation how to write .gatt file by hand would be welcome, we can all go through the Bluetooth documentation to find required services and characteristics.

PS: I can take care of such a task, if you need a little help.

mringwal, 2023-06-05:
Thanks for pointing out that the Bluetooth SIG has eventually removed their nice GATT Service descriptions. In 2020, they've removed the list of these, that's why we added the last list to the script. With the xml files gone, the script has become obsolete.

Please read https://bluekitchen-gmbh.com/btstack/#profiles/#gatt-server and let us know what could be improved, or if how we can make it easier to find it.

mklemarczyk, 2023-06-11:
Hi @mringwal, thank you for the explanation. I think it would be good to remove the script or introduce information how to use it without the original information.

The documentation on the GATT file is better in the link you attached, but still some information are missing.

Some points to specify:
- How to define characteristic description. Like units, additional free text description, etc.
- How implement correctly notify pattern for custom characteristic.
- How to fix the generated index for given characteristic, so it does not change when order of characteristics change in GATT definition file.
- Is there other kinds of services than primary and how to use them ?

nanderLP, 2024-07-30:
A year later and the paragraph with the script is still present, even though it has been deleted for a while now. I was trying to find the convert_gatt_file.py script, until I found this issue. Could someone who is more knowledged on this topic try to rewrite this paragraph?
https://github.com/bluekitchen/btstack/blob/2b49e57bd1fae85ac32ac1f41cdb7c794de335f6/doc/manual/docs-template/profiles.md?plain=1#L649-L693

mringwal, 2024-08-02:
Thanks for the reminder. We've update that paragraph on develop branch to explain the manual process.
https://github.com/bluekitchen/btstack/blob/develop/doc/manual/docs-template/profiles.md#implementing-standard-gatt-services-secgattstandardservices

What service do you want to implement?

closed 2024-08-02 by mringwal

nanderLP, 2024-08-02:
I actually don't need anything in particular right now, I just wanted to try the script while studying the library. The new explanation also makes it easy to recreate a service with given characteristics using a .gatt file, so I'm all good for now. Thank you very much!

mringwal, 2024-08-06:
Glad the new explanation is clear enough for you to move on. Have fun!