bluekitchen/btstack #486: hfp_demo sine wave not working for ESP32

christianfrancouib, 2023-06-12:
Hi everyone, and before hand, thanks for your support. 
**Describe the bug**
I'm trying to run the hfp_demo using the `SCO_DEMO_MODE_SINE` on my ESP-EYE board, but I can't receive any data on my PC. 
The project compiles correctly and I can flash it. Then I can see the HFP device from my PC (BT connection) and can pair it. From the PC side, the device is detected as a Hands-free device successfully. Then I do a test to record the audio coming (I have used different methods: audacity, voice recorder, and Windows microphone testing), but nothing is received. Having a look into the UART console, I see no errors or warnings, the connection is ok, as shown here:

```
Audio connection established with SCO handle 0x0180.
Using mSBC codec.
SCO Demo: Init mSBC
I (34745) I2S: queue free spaces: 2
I (34747) I2S: APLL expected frequency is 8192000 Hz, real frequency is 8191999 Hz
I (34752) I2S: DMA Malloc info, datalen=blocksize=640, dma_buf_count=2
I (34754) I2S: I2S0, MCLK output by GPIO0

Voice recognition ACTIVATED

Summary: data 0100040, packets 2501, packet with crc errors 0, byte errors 0000

Voice Recognition DEACTIVATED

Audio connection released
SCO demo close
SCO demo statistics: Used mSBC with PLC, number of processed frames: 
 - 0 good frames, 
 - 0 zero frames, 
 - 0 bad frames.
I (55697) I2S: DMA queue destroyed
```
My obvious question is if this example is deprecated or if I have to do something else to make it work. 

**To Reproduce**

Steps to reproduce the behavior:
1. Run example 'hfp_demo' with `SCO_DEMO_MODE_SINE` activated.
2. Connect to the device from the PC by BlueTooth and see if the sine wave is captured.

**Expected behavior**

The sine wave is captured by the PC.

**Environment:  (please complete the following information):**
 - Current BTstack branch:  master
 - Bluetooth Controller [ESP32]
 - Remote device: [PC with Windows 10 and 11, tested on both]

mringwal, 2023-06-12:
Hi there. It's supported and should work. The number of frames in your debug output is zero, which is wrong. Looks like SCO (voice audio) is not routed to the stack/demo.

Which version of esp-idf are you using?

Cheers
 Matthias

christianfrancouib, 2023-06-12:
Hi, I've tried. This is the log output, it keeps doing the same.

```
I (275) cpu_start: Pro cpu up.
I (275) cpu_start: Single core mode
I (283) cpu_start: Pro cpu start user code
I (283) cpu_start: cpu freq: 240000000 Hz
I (283) cpu_start: Application information:
I (288) cpu_start: Project name:     hfp_hf_demo
I (294) cpu_start: App version:      6faacb0-dirty
I (299) cpu_start: Compile time:     Jun 12 2023 16:30:04
I (305) cpu_start: ELF file SHA256:  1b69c6215d8cbcd8...
I (311) cpu_start: ESP-IDF:          v5.0.2-dirty
I (316) cpu_start: Min chip rev:     v0.0
I (321) cpu_start: Max chip rev:     v3.99
I (326) cpu_start: Chip rev:         v1.0
I (331) heap_init: Initializing. RAM available for dynamic allocation:
I (338) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (344) heap_init: At 3FFB7468 len 00000B98 (2 KiB): DRAM
I (350) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (356) heap_init: At 3FFD48D8 len 0000B728 (45 KiB): DRAM
I (362) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (369) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (375) heap_init: At 40098608 len 000079F8 (30 KiB): IRAM
I (381) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (389) spi_flash: detected chip: generic
I (392) spi_flash: flash io: dio
W (396) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (409) esp_core_dump_uart: Init core dump to UART
I (415) cpu_start: Starting scheduler on PRO CPU.
I (443) uart: queue free spaces: 10
SDP service record size: 89
Disable BR/EDR Secure Connctions due to incompatibilities with SCO connections
SCO Demo: Sending sine wave, audio output via btstack_audio.
I (458) BTDM_INIT: BT controller compile version [2c56073]
I (462) system_api: Base MAC address is not set
I (471) system_api: read default base MAC address from EFUSE
I (475) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
BTstack up and running at 34:86:5D:52:0F:62
Supported codecs: CVSD, mSBC
Service level connection established 00:E0:3C:3F:9F:03.

AG Indicator Mapping | INDEX 1: range [0, 1], name 'service'
AG Indicator Mapping | INDEX 2: range [0, 1], name 'call'
AG Indicator Mapping | INDEX 3: range [0, 3], name 'callsetup'
AG Indicator Mapping | INDEX 4: range [0, 2], name 'callheld'
AG Indicator Mapping | INDEX 5: range [0, 5], name 'signal'
AG Indicator Mapping | INDEX 6: range [0, 1], name 'roam'
AG Indicator Mapping | INDEX 7: range [0, 5], name 'battchg'
AG Indicator Status  | INDEX 1: status 0x00, 'service'
AG Indicator Status  | INDEX 2: status 0x00, 'call'
AG Indicator Status  | INDEX 3: status 0x00, 'callsetup'
AG Indicator Status  | INDEX 4: status 0x00, 'callheld'
AG Indicator Status  | INDEX 5: status 0x00, 'signal'
AG Indicator Status  | INDEX 6: status 0x00, 'roam'
AG Indicator Status  | INDEX 7: status 0x01, 'battchg'
Speaker volume: gain 9
Microphone volume: gain 9
AG Indicator Status  | INDEX 3: status 0x00, 'callsetup'
Audio connection established with SCO handle 0x0180.
Using mSBC codec.
SCO Demo: Init mSBC
W (139128) i2s_common: APLL is occupied already, it is working at 8191999 Hz while the expected frequency is 8192000 Hz
W (139131) i2s_common: Trying to work at 8191999 Hz...

Voice recognition ACTIVATED

Summary: data 0100040, packets 2501, packet with crc errors 0, byte errors 0000
Summary: data 0100040, packets 2501, packet with crc errors 0, byte errors 0000

Voice Recognition DEACTIVATED

Audio connection released
SCO demo close
SCO demo statistics: Used mSBC with PLC, number of processed frames:
 - 0 good frames,
 - 0 zero frames,
 - 0 bad frames.
 ```

mringwal, 2023-06-12:
Hm.. the summary lines mentions 2.5k packets, but the SCO statistics show 0 frames.
Could you uncomment the call to hci_dump_init in main/main.c and post the log, too, please?

christianfrancouib, 2023-06-13:
I don't know if it may help, but I have uncommented this line you mentioned to see what happen and this is what is shown when I pair with it, ask for the samples (pressing to record the voice from the Audacity app), let it for a minute "recording" and then stop it:

```
Audio connection established with SCO handle 0x0180.
Using mSBC codec.
SCO Demo: Init mSBC
W (98796) i2s_common: APLL is occupied already, it is working at 8191999 Hz while the expected frequency is 8192000 Hz
W (98803) i2s_common: Trying to work at 8191999 Hz...
[00:01:38.813] LOG -- btstack_audio_esp32.c.345: i2s init samplerate 16000, samples per DMA buffer: 160

Voice recognition ACTIVATED

[00:01:38.824] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:01:38.842] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:01:38.853] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:01:38.864] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:01:38.874] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:01:38.893] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:01:38.904] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:01:38.915] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:01:38.934] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:01:38.945] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:01:38.956] EVT <= 6E 00
[00:01:38.956] EVT <= 6F 00
[00:01:38.966] SCO => 80 01 3C 01 08 AD 00 00 C1 E9 97 67 66 7F 77 6D 60 1D DB 58 48 86 D6 32 60 C4 AF BB A0 7E 42 28 6E 48 76 D6 29 DD B5 27 77 6D 40 1D DB 52 C7 76 D6 41 DD B5 E8 77 6D 7E DD DB 5B C7 76 D4 00
[00:01:38.977] EVT <= 6E 00
[00:01:38.985] EVT <= 6F 00
[00:01:38.986] SCO => 80 01 3C 01 38 AD 00 00 F4 E1 21 11 00 55 32 C8 82 6D B4 20 C7 62 35 6E 34 B4 C2 C8 F6 3F 71 47 4E 56 9C B6 28 3A 02 27 D2 C8 80 05 A6 62 CD 1A C6 42 23 D8 E9 64 EE 3F 11 55 1B D0 A4 98 00
[00:01:39.006] EVT <= 6E 00
[00:01:39.007] EVT <= 6F 00
[00:01:39.008] SCO => 80 01 3C 01 C8 AD 00 00 F4 E1 21 11 00 55 32 C8 82 6D B4 20 C7 62 35 6E 34 B4 C2 C8 F6 3F 71 47 4E 56 9C B6 28 3A 02 27 D2 C8 80 05 A6 62 CD 1A C6 42 23 D8 E9 64 EE 3F 11 55 1B D0 A4 98 00
[00:01:39.027] EVT <= 6E 00
[00:01:39.028] EVT <= 6F 00
[00:01:39.036] SCO => 80 01 3C 01 F8 AD 00 00 F4 E1 21 11 00 55 32 C8 82 6D B4 20 C7 62 35 6E 34 B4 C2 C8 F6 3F 71 47 4E 56 9C B6 28 3A 02 27 D2 C8 80 05 A6 62 CD 1A C6 42 23 D8 E9 64 EE 3F 11 55 1B D0 A4 98 00
[00:01:39.048] EVT <= 6E 00
[00:01:39.057] CMD => 35 0C 09 02 80 01 0A 00 80 00 01 00
[00:01:39.059] EVT <= 6E 00
W (99061) AUDIO: audio output buffer underrun
W (99067) AUDIO: audio output buffer underrun
[00:02:41.952] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:02:41.958] CMD => 35 0C 05 01 80 01 01 00
[00:02:41.960] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:02:41.968] SCO <= 80 21 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:02:41.979] ACL <= 80 20 15 00 11 00 41 00 0B FF 19 00 0D 0A 2B 42 56 52 41 3A 20 30 0D 0A 86
[00:02:41.989] LOG -- rfcomm.c.1447: RFCOMM data UIH_PF, new credits channel 0x01: 0, now 32
[00:02:41.998] LOG -- hfp.c.2144: HFP_HF_RX: '\r\n+BVRA: 0\r\n'
[00:02:42.008] LOG -- hfp.c.1394: command string '+BVRA:', handsfree 1 -> cmd id 37
[00:02:42.010] EVT <= 05 04 00 80 01 13
Audio connection released
SCO demo close
SCO demo statistics: Used mSBC with PLC, number of processed frames:
 - 0 good frames,
 - 0 zero frames,
 - 0 bad frames.
[00:02:42.029] LOG -- btstack_audio_esp32.c.365: i2s close
[00:02:42.039] LOG -- hci.c.1282: Connection closed: handle 0x180, 00:E0:3C:3F:9F:03
[00:02:42.041] LOG -- hci.c.7753: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[00:02:42.049] EVT <= 61 01 01
[00:02:42.050] EVT <= 1B 03 80 00 05
[00:02:42.058] EVT <= 6E 00
[00:02:42.059] EVT <= 6F 00

Voice Recognition DEACTIVATED

[00:02:42.062] CMD => 35 0C 05 01 80 00 01 00
[00:02:42.069] EVT <= 6E 00
```
if you have a look, there's a warning saying: `W (99061) AUDIO: audio output buffer underrun`
Hope it helps, thank you.

mringwal, 2023-06-13:
Please upload the full log from RESET as a text file/zip. With log enabled, audio won't work, but that's expected.

mringwal, 2023-06-13:
thanks. The received SCO have an weird length, they should 60 bytes for mSBC but they are 40. Even if the incoming data is somehow discarded, it doesn't explain why you don't receive anything.
Please try CVSD as well by adding a #undef ENABLE_HFP_WIDE_BAND_SPEECH after the #include <..> lines in the hfp_hf_demo. Without HCI dump to see/hear if it works for you  and post the version HCI dump here for a quick check.

I'll need to setup an ESP32 and test myself to understand this better.

mringwal, 2023-06-13:
Thanks. I'll re-test HFP on ESP32 soon. I did test HFP HF on my Mac with an USB dongle last week for the release and that was working as expected.

christianfrancouib, 2023-06-13:
Thank you very much !

mringwal, 2023-06-14:
Please try again with the current version on master or develop. Should work now as expected.

christianfrancouib, 2023-06-14:
Hi Mathias,
Firstly, I want to thank you once again for your time and quick answers. I greatly appreciate it.
I have tested the new repository again, and now I can see the light 😊... However, I face some issues: Let me explain my experiments...

### 1. Tests without HCI (hci line commented as default)
The first thing I did was to test it as it comes by default, only changing the mode to sine_wave. The result was that when I tried to access the microphone from my PC, I received the sine wave as expected. But I face some issues:

a. At the end of the log, the system once again presented `0 good frames` even though it was successfully sending the data.

b. The second issue occurred after making the first recording, stopping it, waiting for approximately 20 seconds, and then attempting to record again. The result was that I couldn't receive any sound from that point onwards (similar to the original problem). To resolve this, I had to unpair and re-pair the devices, after which it worked again. I believe this issue is time-related because if I record and stop repeatedly within a few seconds, it always works fine. The problem arises when I let it remain idle for several seconds.

I have attached the log documenting these results as `log_without_hci_first_ok_then_after_time_fail.txt`.

### 2. Tests with HCI activated

What I did next was to test it with the HCI line activated. The result was the same. It worked (using CVSD, I noticed you changed this by default) if I didn't let it stay idle for too long, but it would break when there were a few seconds of inactivity. I have included the log when it works, labeled as `log_with_hci_ok`, and another log showing the issue after a period of inactivity, named `log_with_hci_failed_after_time`.

When the HCI is activated, I observed that after a few seconds of inactivity, the logging displays the following message:

```
[00:00:51.192] EVT <= 14 06 00 80 00 02 40 06
[00:00:51.195] LOG -- hci.c.3841: HCI_EVENT_MODE_CHANGE, handle 0x0080, mode 2
```
When these lines appear, it stops working (you can see this in the initial lines of the `log_with_hci_failed_after_time` file). 

Hopefully, this can help. 

Thank you again for your great support. 

[log_without_hci_first_ok_then_after_time_fail.txt](https://github.com/bluekitchen/btstack/files/11747518/log_without_hci_first_ok_then_after_time_fail.txt)

[log_with_hci_ok.txt](https://github.com/bluekitchen/btstack/files/11747801/log_with_hci_ok.txt)

[log_with_hci_failed_after_time.txt](https://github.com/bluekitchen/btstack/files/11747802/log_with_hci_failed_after_time.txt)

mringwal, 2023-06-14:
Thanks for the feedback. At the moment, I'm clueless why the number of processed frames is zero: these numbers are for received frames not for sent frames, btw. I didn't check on the esp32 (which is already stowed away..)

The other irritating this is why CVSD was used the second time, but here, the other side might chose to do so.

Instead of testing with the built-in Windows Bluetooth, you could run the hfp_ag_demo with an USB Bluetooth Dongle, which provide you with packet logs from the HF side (HCI logging on the desktop does not stop audio).

Anyway, we have fixed another 'cosmetic' detail, which is worth giving a try.

christianfrancouib, 2023-06-15:
Hi again Mathias,
I have conducted another experiment that could shed some light on the issue. I have reverted to an older commit of the `btstack` project, specifically the one from August 4, 2021 (commit `6320fbc8`). I did this because I was attempting to follow an example project created by an individual using your `btstack`library. Since his project was from July 2021, I chose this particular date.

I followed the necessary steps to update the `btstack` repository and utilized the `hfp_demo` example from that specific date. To my delight, it worked flawlessly. Even when leaving the device idle for several minutes, the connection remained stable and the recording function operated as expected.

As usual, I have included the log files for this scenario, both with HCI and without it.

Let me know whatever you can find.

Thank you again

[log_without_hci_commit_old_commit_6320fbc8.txt](https://github.com/bluekitchen/btstack/files/11755706/log_without_hci_commit_old_commit_6320fbc8.txt)
[log_with_hci_commit_old_commit_6320fbc8.txt](https://github.com/bluekitchen/btstack/files/11755707/log_with_hci_commit_old_commit_6320fbc8.txt)

mringwal, 2023-06-15:
Thanks, but while this shows that your setup (ESP32 + PC) is working in principle, it doesn't help with the issue at hand.
Did you try the current version again? If it fails, please also try to just re-connect without first unpair/pair.

Btw. any idea why CVSD was used? Could this be related to the way you've initiated the audio connection?

christianfrancouib, 2023-06-16:
Hi Mathias, 
yes I have tried the current version and the problem persists. For getting it to work again after recording for the first time, I do a re-connect process, not unpair/pair. 
Regarding CVSD, I have no idea. I thought it was something you changed in the library.
Please let me know if I have to do some other experiment to help you out.

mringwal, 2023-06-17:
Hi. Could you try with other devices, e.g. a smartphone, preferably an iOS device? I've tried the Voice Recorder on my iPhone and it worked well, even with longer pauses while still being connected.

mringwal, 2023-06-19:
No, I didn't. Just wanted to have a reference point. Could you now try with Android and Windows for comparison?

christianfrancouib, 2023-06-20:
Hi Mathias, I have tried with W10, and W11. For the case of W10, everything worked ok, but in the W11 case, I still face the time issue. I don't have an Android device so I cant test with it yet.

mringwal, 2023-06-20:
Ok, more data. So, we got iOS & Windows 10: ok, Windows 11: fail. 

Can you run the HFP HF demo on a desktop system to get nice logs?
If not, please send another log from the ESP32 with HCI Logging enabled.

christianfrancouib, 2023-06-20:
For the W11 failure case you mean? I have done it before in this entrance: https://github.com/bluekitchen/btstack/issues/486#issuecomment-1591488567
That was for the W11 case with and without HCI.

mringwal, 2023-06-22:
HI @christianfrancouib In that log, BTstack tried to setup an SCO connection after accepting an incoming SCO connection failed, which is wrong. I have streamlined the logic for that.

Could you update to the current version of develop and try again? If it fails, please send a complete HCI log from the start, i.e. including the working and the non-working audio setup.

christianfrancouib, 2023-06-22:
Ok Mathias, I will test it again in the W11 machine.

christianfrancouib, 2023-06-23:
By the way Mathias, I don't know if I need to open a new thread for this, but don't you have some documentation or a template example or something I can take as a guide for using the example with a microphone. Thank you again.

mringwal, 2023-06-23:
@christianfrancouib Please post general questions on the community mailing list, see readme.md

The microphone should work on the ESP32 LyraT v4.3 dev kit. 
You can use the audio_duplex example to verify that your audio drivers are working properly before trying with HFP HF.

christianfrancouib, 2023-06-23:
Where is this community mailing list?...

mringwal, 2023-06-23:
**Discussion and Community Support:** [BTstack Google Group](https://groups.google.com/group/btstack-dev)

christianfrancouib, 2023-06-24:
Hi Mathias, finally I could make it work with the microphone in my ESP-EYE board. Everything is working fine and no issues have been observed in W10 machine and iOS mobile.

mringwal, 2023-07-06:
Great.

closed 2023-07-06 by mringwal