bluekitchen/btstack #487: A2DP sink on ESP32

0x0fe, 2023-06-15:
Is there a runnable project (arduino, pio or IDF) for evaluating quickly the bluekitchen bt stack with ESP32 on A2DP sink (to I2S) mode? Also i would like to know in which measure this stack will replace the IDF stack, in particular, how much ressources (IRAM) are freed when using it i dont need a number, but rather a confirmation that it will use significantly less IRAM). 

The reason i want to evaluate this is because my device has to run simultaneously wifi STA, https stream, decoding and playback of MP3 files, https download of mp3 files, reading/decoding/playback of mp3 from sd_mmc, and other basic functions such as adkeys and assignable leds. 
With IDF it requires a lot of modifications in the sdkconfig to be able to run all this and not overflow the IRAM (i used esp32d0wdr2 which has 2MB PPSRAM too, but the PSRAM cannot be used for most of the very memory hungry SDK functions, in particular the wifi and bluetooth stack just use few tens of KB from the PSRAM and most of the IRAM available. 
My main problem is that many of these modification impede overall performances significantly, in particular download speed via wifi, and also the very small amount of remaiing IRAM even after these massive modifications of the sdkconfig lead to the sensation of really being on the edge in term of ressource usage.

Of course, i never run A2DP and mp3 download or mp3 streaming/playback at the same time, the A2DP is used exclusively of  other sources and i "could" free ressources when not used, however there is a major flaw in the espressif bluetooth stack implementation, we cannot stop it and free ressources, if we stop and free ressources we cannot restart it, we can only stop it without freeing ressources, which is very silly, but documented and confirmed.  Also i consider the amount of ressources used by the espressif bluetooth stack to be excessive, when both bluetooth (A2DP) and wifi are enabled more than 3/4 of the IRAM are used, which is just too much.

mringwal, 2023-06-16:
Hi there. With BTstack on the ESP32, the Bluedroid stack gets replaced. It has multiple layers of abstraction and uses dynamic memory and buffers liberally, in contrast to BTstack which has a very slim abstraction layer and minimizes memory use.

I have no numbers as I never tried Bluedroid on the ESP32 only our stack. Just follow the instructions in port/esp32 and compare to your existing numbers for a similar setup (without wifi etc..).

closed 2023-06-16 by mringwal

0x0fe, 2023-06-16:
thank you for the reply. So i should disable the bluetooth in menuconfig? (i use IDF, but i have the Arduino HAL in the project)

mringwal, 2023-06-16:
if you start with our example projects, Bluedroid is disabled by default and there are warnings for some common configuration error cases.

0x0fe, 2023-06-16:
ok noted