bluekitchen/btstack #488: this is for a firmware that uses your code. i need help

TheRealKush1, 2023-06-17:
so im not good at C code but im starting to figure out how it works a little bit. from what ive gathered about this firmware(Blue Cube Mod) these lines that say include btstack.h are refering to a header file called that but i cant find that header file in the files. has it been renamed or somethig? heres a snippet of the code that im talking about. ime fairly new to coding and have only ever used python so any help you could offer would be appreciated. ive been trying to figure it out for over a week now and cant get the stupid firmware flashed and dude just fell off the face of the earth 2 years ago. 

#include "btstack.h"
#include "btstack_event.h"
#include "btstack_stdin.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "freertos/semphr.h"
#include "btstack_run_loop_freertos.h"

mringwal, 2023-06-18:
Please post on the BTstack community mailing list and/or from the project you're trying to use.

'btstack.h' is at btstack/src/btstack.h

I would suggest to play with our examples in a btstack/port that is close to your target

mringwal, 2023-06-18:
Not a bug

closed 2023-06-18 by mringwal