bluekitchen/btstack #489: hid_mouse_demo.c: insufficient buffer size

iAmGroute, 2023-06-18:
In this example project, the `hid_service_buffer` has a size of 250 bytes:  
https://github.com/bluekitchen/btstack/blob/bedb5085b2a5e2ca2ef2d5b85a1ab64e543f2fde/example/hid_mouse_demo.c#L65

But, according to the printf message, it needs to be at least 254:
https://github.com/bluekitchen/btstack/blob/bedb5085b2a5e2ca2ef2d5b85a1ab64e543f2fde/example/hid_mouse_demo.c#L326-L328

```
SDP service record size: 254
```

I was (un)lucky enough to catch this, because a variable from [a different repo](https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/rp2_common/pico_cyw43_driver/cybt_shared_bus/cybt_shared_bus_driver.c#L66) was placed right next to it in bss and was getting overwritten with inexplicable values.

Maybe adding an assertion to check for this would be a good idea, perhaps in `hid_create_sdp_record()`.

Btw, I love the variety of these examples, great work!

mringwal, 2023-06-19:
Thanks for reporting and sorry for this. We'll increase the buffer there and also add asserts to at least check to abort after overwriting memory.

mringwal, 2023-07-30:
Fixed on develop branch. Closing it here.

commit 762141af (2023-06-20): example: use sdp_create_service_record_handle() for sdp record and assert sdp record buffer is large enough, fixes #489
closed 2023-07-30 by mringwal