bluekitchen/btstack #492: Add support for HCI_SUBEVENT_LE_ENHANCED_CONNECTION_COMPLETE_V1

amorniroli, 2023-06-21:
**Is your feature request related to a problem? Please describe.**
Some HCI controllers (e.g. ST BLUENRG-M2SP), in HCI_SUBEVENT_LE_CONNECTION_COMPLETE event are not reporting peer address correctly. In my case, even if resolution address is enabled, the peer address is reported with random address instead of it's public address. This is causing some issues in hci_connection_for_bd_addr_and_type since two connections would be created.

**Describe the solution you'd like**
HCI_SUBEVENT_LE_ENHANCED_CONNECTION_COMPLETE_V1 could be used to avoid this issue, since this event packet would contain both resolved and random address.

I've made some changes starting from 249b703b04d67c7b50dac2cf26bd62a06ecf703d and everything seems to work as expected (at least in my use case scenario).

Attached a patch containing the changes.

[patch.zip](https://github.com/bluekitchen/btstack/files/11818519/patch.zip)

**Describe alternatives you've considered**
/

**Additional context**
The feature could be added to btstack_config.h file (e.g. ENABLE_ENHANCED_CONNECTION_COMPLETE) in order to select valid mask for  HCI_INIT_W4_LE_SET_EVENT_MASK.

mringwal, 2023-06-21:
Hi. Thanks for the suggestion and the patch. We have a related TODO open. Updating the core stack to handle HCI_SUBEVENT_LE_ENHANCED_CONNECTION_COMPLETE_V1 is the easier part, we're unsure what to do about application code that expects the original connection complete event.

I guess a separate define like the ENABLE_ENHANCED_CONNECTION_COMPLETE you've suggested might be needed.

mringwal, 2023-06-22:
We did discuss this. To avoid breaking backwards-compatibility. we'll use a ENABLE_HCI_ENHANCED_CONNECTION_COMPLETE define - btw. there's also a HCI_SUBEVENT_LE_ENHANCED_CONNECTION_COMPLETE_V2.

To make it easier to use, we'll introduce a generic gap_le_connection_complete event with the fields from conn complete v2 and boolean flags to indicate if additional attributes are available. The examples can then switch to that event without additional #ifdef + handling all three connection complete variants.

I'll review your patch and commit it this week, the new event will come later.

mringwal, 2023-11-27:
We have added a new generic GAP_SUBEVENT_LE_CONNECTION_COMPLETE that's matches the HCI_SUBEVENT_LE_ENHANCED_CONNECTION_COMPLETE_V2 but it is constructed from all three possible connection events. (We also updated all higher layers to use the new event instead of adding all three connection events).

To enable the new events, ENABLE_LE_ENHANCED_CONNECTION_COMPLETE_EVENT needs to be added to btstack_config.h. 

Could you give this a quick try? We want to prepare a new BTstack release and it would be great to know that this also works for your case.

amorniroli, 2023-12-02:
Hi @mringwal, that's a good news!

I should be able to test it tuesday: I'll let you know.

Alessandro

mringwal, 2023-12-19:
Hi again. While the additional GAP_SUBEVENT_LE_CONNECTION_COMPLETE looks rather solid to me, the ENABLE_LE_ENHANCED_CONNECTION_COMPLETE_EVENT make me wonder where the additional addr types (2 & 3) are used and if that's correct. 

Is there any need to receive HCI_SUBEVENT_LE_ENHANCED_CONNECTION_COMPLETE_V1 or HCI_SUBEVENT_LE_ENHANCED_CONNECTION_COMPLETE_V2? If yes, please let me know.

For now, I would recommend to not enable ENABLE_LE_ENHANCED_CONNECTION_COMPLETE_EVENT. We will do more reviews and let you know.

mringwal, 2024-03-07:
Closing, it's properly handled and used as far as I can tell. Please re-open if you run into an issue with this.

closed 2024-03-07 by mringwal