bluekitchen/btstack #497: ESP32 GATT over BR/EDR cannot connect to iPhone

Omegaki113r, 2023-07-09:
I tried almost all the examples provided. All of them Tries to connect with IPhone and become unsuccessful. 

The message that appears on the screen is, 
"Connection Unsuccessful
Make sure <DEVICE NAME I SET> is turned on and in range"

I have put,
#define ENABLE_LOG_DEBUG
#define ENABLE_LOG_ERROR
#define ENABLE_GATT_OVER_CLASSIC

on the top of the files as well.

I will attach the LOG i see in the terminal too. 

`[00:00:00.479] LOG -- btstack_run_loop_freertos.c.236: run loop init, task 0x3ffbd63c, queue item size 8
[00:00:00.489] LOG -- main.c.283: transport_register_packet_handler
[00:00:00.519] LOG -- btstack_tlv.c.48: TLV Instance 0x3f4099d4
[00:00:00.519] LOG -- le_device_db_tlv.c.159: num valid le device entries 0
[00:00:00.529] LOG -- l2cap.c.4796: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
SDP service record size: 58
[00:00:00.529] LOG -- att_db.c.237: att_set_db 0x3f40393c
[00:00:00.529] LOG -- hci.c.1612: Num LE Peripheral roles: 0 of 1
[00:00:00.539] LOG -- hci.c.5207: hci_power_control: 1, current mode 0
[00:00:00.539] LOG -- main.c.203: transport_init
[00:00:00.549] LOG -- main.c.214: transport_open
I (549) BTDM_INIT: BT controller compile version [6a07b06]
I (559) system_api: Base MAC address is not set
I (559) system_api: read default base MAC address from EFUSE
I (569) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (579) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
[00:00:01.219] LOG -- hci.c.7684: BTSTACK_EVENT_STATE 1
[00:00:01.229] EVT <= 60 01 01 
[00:00:01.229] CMD => 03 0C 00
[00:00:01.239] EVT <= 0E 04 05 03 0C 00
[00:00:01.239] EVT <= 6E 00
[00:00:01.239] CMD => 01 10 00
[00:00:01.249] EVT <= 0E 0C 05 01 10 00 08 0E 03 08 60 00 0E 03 
[00:00:01.249] LOG -- hci.c.2849: Manufacturer: 0x0060
[00:00:01.249] EVT <= 6E 00
[00:00:01.259] CMD => 14 0C 00
[00:00:01.259] EVT <= 0E FC 05 14 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:01.319] LOG -- hci.c.2695: local name: 
[00:00:01.319] EVT <= 6E 00 
[00:00:01.329] CMD => 02 10 00 
[00:00:01.329] EVT <= 0E 44 05 02 10 00 BF FF FF 03 CC FF EF FF 3F FF FC 1F F2 0F E8 FE 3F F7 8F FF 1C 00 04 00 61 F7 FF FF 7F 38 C0 FF 7F FF FF 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:01.349] LOG -- hci.c.2663: Command 0x00 supported 2/5
[00:00:01.349] LOG -- hci.c.2663: Command 0x01 supported 10/4
[00:00:01.359] LOG -- hci.c.2663: Command 0x02 supported 14/7
[00:00:01.359] LOG -- hci.c.2663: Command 0x03 supported 18/3
[00:00:01.369] LOG -- hci.c.2663: Command 0x04 supported 20/4
[00:00:01.379] LOG -- hci.c.2663: Command 0x05 supported 22/2
[00:00:01.379] LOG -- hci.c.2663: Command 0x06 supported 24/6
[00:00:01.389] LOG -- hci.c.2663: Command 0x07 supported 32/1
[00:00:01.389] LOG -- hci.c.2663: Command 0x08 supported 32/3
[00:00:01.399] LOG -- hci.c.2663: Command 0x09 supported 32/6
[00:00:01.399] LOG -- hci.c.2663: Command 0x0a supported 34/0
[00:00:01.409] LOG -- hci.c.2663: Command 0x0b supported 35/1
[00:00:01.409] LOG -- hci.c.2663: Command 0x0c supported 35/3
[00:00:01.419] LOG -- hci.c.2668: Local supported commands summary 00001fff
[00:00:01.419] LOG -- btstack_crypto.c.1121: controller supports ECDH operation: 1
[00:00:01.429] EVT <= 6E 00
[00:00:01.429] CMD => 09 10 00
[00:00:01.439] EVT <= 0E 0A 05 09 10 00 1E 91 5F A4 AE 30
[00:00:01.439] LOG -- hci.c.2792: Local Address, Status: 0x00: Addr: 30:AE:A4:5F:91:1E
[00:00:01.449] EVT <= 6E 00
[00:00:01.449] CMD => 05 10 00 
[00:00:01.459] EVT <= 0E 0B 05 05 10 00 FD 03 FF 09 00 04 00 
[00:00:01.459] LOG -- hci.c.2712: hci_read_buffer_size: ACL size module 1021 -> used 1021, count 9 / SCO size 255, count 4
[00:00:01.469] EVT <= 6E 00
[... 337 lines cut ...]
[00:00:25.489] EVT <= 6E 00
[00:00:25.499] LOG -- btstack_crypto.c.956: more ecc random
[00:00:25.499] CMD => 18 20 00
[00:00:25.509] EVT <= 0E 0C 05 18 20 00 64 FF E0 A8 14 FE B3 D8
[00:00:25.509] EVT <= 6E 00
[00:00:25.509] LOG -- btstack_crypto.c.956: more ecc random
[00:00:25.519] CMD => 18 20 00 
[00:00:25.519] EVT <= 0E 0C 05 18 20 00 22 EA 8B D7 C6 0B 15 51
[00:00:25.529] EVT <= 6E 00
[00:00:25.529] LOG -- btstack_crypto.c.956: more ecc random
[00:00:25.529] CMD => 18 20 00
[00:00:25.539] EVT <= 0E 0C 05 18 20 00 BF 9E 0A 92 CF 91 52 9E
[00:00:25.539] LOG -- btstack_crypto.c.525: set uECC RNG for initial key generation with 64 random bytes
[00:00:25.549] LOG -- btstack_crypto.c.498: sm_generate_f_rng: size 32 - offset 0
[00:00:25.719] EVT <= 6E 00 
[00:00:25.719] LOG -- btstack_crypto.c.488: Elliptic curve: X
[00:00:25.729] LOG -- btstack_util.c.274: 0x44, 0x57, 0x24, 0x65, 0x3C, 0xE1, 0xF0, 0x28, 0x96, 0xD6, 0x0C, 0x03, 0xD1, 0xDE, 0x55, 0xF4, 
[00:00:25.729] LOG -- btstack_util.c.274: 0xF0, 0x06, 0x3D, 0x80, 0x9F, 0x1D, 0xE7, 0xC7, 0x30, 0x9A, 0x64, 0x21, 0x6F, 0x42, 0xBF, 0xED,
[00:00:25.739] LOG -- btstack_crypto.c.490: Elliptic curve: Y
[00:00:25.739] LOG -- btstack_util.c.274: 0xC8, 0x7D, 0x7D, 0x98, 0xAF, 0xC2, 0x79, 0xC1, 0x77, 0x68, 0xF0, 0xA6, 0xD2, 0x1E, 0x80, 0x3B,
[00:00:25.749] LOG -- btstack_util.c.274: 0x8D, 0xCE, 0x8F, 0x7D, 0x81, 0x72, 0x3F, 0x18, 0x62, 0x47, 0xEB, 0xD1, 0x0A, 0xCE, 0xCF, 0xA0, `

Additional information:
If i try the pan_lwip_http_server.c example. everything works correctly and the iPhone indeed get connected to the ESP32 successfully. 
But what i need is a BR/EDR connection over GATT to IPhone to stream some RAW audio data so i can process them on the iPhone.

mringwal, 2023-07-10:
hi. pretty much all examples should work on the esp32. 
To enable GATT over Classic on the EPS32, you need to add ENABLE_GATT_OVER_CLASSIC to btstack_config.h in esp-idf/components/btstack/include, or update it in port/esp32/components/btstack/include before you call ./integrate_btstack.h

With that, you can run gatt_streamer_server, which support GAT over LE as well as BR/EDR.

Now, the question is how to make iOS use it. If I use iOS Bluetooth Settings, the gatt_streamer_server doesn't show up. If I cheat and set the CoD to HFP, I get "Pairing Unsuccessful". 

How did you test this?

Omegaki113r, 2023-07-10:
Hi.. Thank you for the response.

"If I use iOS Bluetooth Settings, the gatt_streamer_server doesn't show up" => in my case it does show up. But i had to restart my iPhone 1 time and then scan inside the Settings to make it appear. But when i try to connect it failed. I mean it seems to have paired with the iPhone. (Because the device appeared inside "MY DEVICES" list) but i was never able to connect to it. it always said to check whether the device is turned ON or in range. 

"you need to add ENABLE_GATT_OVER_CLASSIC to btstack_config.h" => i didn't knew about this. i only added "ENABLE_GATT_OVER_CLASSIC " at the top of my main application code. 

Good news is.. After i did above thing. "ENABLE_GATT_OVER_CLASSIC on the btstack_config.h" the ESP32 was successfully connected.. 
Thank you for guiding me in the correct path. 

Though the device get automatically disconnected after like 10 seconds

mringwal, 2023-07-11:
Yes, you can convert the log output with tool/create_packet_log.py and view it in Wireshark. The EVT <= 05 ... is a disconnected event and it does not look like BTstack did trigger the disconnect.

Again, I don't know how GATT over BR/EDR is used on the iOS side. There are a few WWDC videos when support for GATT over BR/EDR was introduced for CoreBluetooth. Please watch these.

mringwal, 2023-07-12:
> Is there a documentation or a header file that describe those event_ids? 
That would be the Bluetooth Core Specification v5.4 - see the Chapter on HCI.

Also: you can convert the log output with tool/create_packet_log.py and view it in Wireshark.

mringwal, 2023-07-12:
Btw. is it correct that the connection is initiated on the iOS side from Settings->Bluetooth?

Omegaki113r, 2023-07-15:
Additional information about the disconnect i mentioned in previous comments.. 

i am still trying to figure out where the issue is coming from (whether it's from the GATT over BR/EDR implementation of the btstack repo.. OR is it a instability of the iOS itself. because i've seen lots of forum threads talking about GATT over BR/EDR not being stable. though those posts are like 1-2 years old now.)

anyways. i have a bit more new information about this disconnection. 

let's assume that the ESP32 i am using is never paired with iPhone before. 

1) i upload the gatt_streamer_server example to ESP32 with the ENABLE_GATT_OVER_CLASSIC  in bt_config
2) open up settings -> bluetooth and try to connect. sometimes it connects without any problems. but sometimes the first try fails. but the 2nd try get successfully connected. 
3) this connection can last upto a day without any disconnections (i tried up to 36 hours esp32 was connected to iPhone)
4) then let's say i re upload the same code to ESP32 without any changes to the code. (Also while the ESP32 connected to iPhone).
5) This reflash of the firmware disconnect the ESP32 from the iPhone (which is expected. )
6) once the flash is done. and when i try to connect the ESP32 back to Iphone again. it takes sometimes unsuccessful attempts before it get connected. even if it gets connected. the connection do not last more than 10 seconds. iPhone disconnects from the ESP32. 
7) No matter how many times i try to reflash the ESP32 it won't connect. 

8) the only way to make the ESP32 connect back to iPhone is, i need to run idf.py -p XXXX erase_flash and then reflash the ESP32 with the same code. which makes the iPhone request from user to repair the ESP32. after that repairing. then the ESP32 again can stay connected to IPhone many many hours. 

9) So if i flash the ESP32 while the ESP32 is connected to iphone it breaks the whole things. (it maybe an authentication thing. not sure yet). 

So to reflash the ESP32. i need to manually disconnect the ESP32 from the iPhone using settings -> bluetooth 
and then reflash ESP32 an connect to iPhone again to test the changes i made in code. 

i just wanted to share my findings. Will update if i have any more news or if i figure out what is causing this. 
If you have an idea what could be the reason please let me know too.

mringwal, 2023-07-16:
Thanks for the update. I've re-tested gatt_counter with a CYW20704 USB dongle which supports BR/EDR Secure Connections and ENABLE_GATT_OVER_CLASSIC. I did pair on the first try and I've verified the HCI log to see if an L2CAP Connection for ATT has opened. It also did trigger Cross-Transport-Key-Deriviation for the LE LTK, without an LE connection.

I've then started an BLE Browser (BLExplorer) on my iPhone which showed "LE Counter' - probably as the gatt counter demo was still advertising. Anyway, I was able to connect and subscribe to notifications. In the HCI log, there was no LE Connection and everything was sent over the Classic connection.

Could you focus on your application to see if you can talk to the ESP32 over GATT/Classic? 
If you have problems to re-connect, please try to delete the pairing on iOS, you should not need to re-flash or erase the ESP32 flash.

mringwal, 2023-09-18:
Closing due to inactivity, works for me.

closed 2023-09-18 by mringwal