bluekitchen/btstack #498: Raspberry Pico W can't store link key

wasdwasd0105, 2023-07-11:
**Describe the bug**
Hello,
I am writing a BTstack project with Pico W and want to save paired info. However, it seems BTstack may not utilize Pico's storage to save link key. gap_store_link_key_for_bd_addr was called to save link key. However, I get nothing when I use list_link_keys() to get the key

**To Reproduce**

Here is my project: https://github.com/wasdwasd0105/PicoW-usb2bt-audio

[image]

**Expected behavior**

A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

mringwal, 2023-07-11:
Hi. That's weird. BTstack uses it's btstack_tlv.h for persistent storage. On the Pico W, this is implemented in btstack_flash_bank.c, which provides pico_flash_bank_instance().

If you run the a2dp_sink_demo and connect from a mobile phone. Can you connect again after the pairing?

I would assume yes - I've played with the Cosmic Unicorn and BTstack a while ago and don't remember the need to pair every time.

@peterharperuk the list_link_keys() functions uses the link key iterator provided by our btstack_link_key_db_tlv.c, which only requires that btstack_tlv is properly set-up. The iteration uses the same logic as the regular look-up. (just to get you into the loop, no action item yet :)

peterharperuk, 2023-07-11:
Yes, it should work. Pico W uses the last two blocks in flash to store the TLV stuff. It's possible to overwrite those, but it doesn't seem likely. Where your code that calls list_link_keys?

mringwal, 2023-07-11:
@peterharperuk if you like, please copy list_link_keys from https://github.com/bluekitchen/btstack/blob/master/example/gap_link_keys.c#L64
into one of the classic examples and call it after pairing (or on start after a reboot..)

wasdwasd0105, 2023-07-15:
[image]

I think I got the reason: the storage system works well if I compile directly from https://github.com/raspberrypi/pico-examples/ and it will fail if I use a third-party temple such as https://github.com/MrGreensWorkshop/RasPiPicoSDK_BT_Classic_SPP. So what should I do to initiate the storage/DB part properly?

wasdwasd0105, 2023-07-15:
And I found if I use a temple project, I can read the data but cant storage data

mringwal, 2023-07-16:
Interesting find. We've only checked our examples on the Pico W. 
BTstack's TLV is initialized from cyw43_arch_init() which is also called from the main.c in MrGreensWorkshop. 

Could you start your app development from one of them? For SPP, please have a look at spp_counter or spp_streamer - the first one sends a counter, the second tries to send as fast as possible.

sunnyqeen, 2023-07-26:
@wasdwasd0105
bool flash_safe_execute_core_init(void) is needed for multi-core
see  pico-sdk\src\rp2_common\pico_flash\include\pico\flash.h
I got similar problem.

wasdwasd0105, 2023-07-28:
Yes! I solve the issue. I tried flash_safe_execute_core_init(void) on core 1 but still returned error -4. But I defined PICO_FLASH_ASSUME_CORE1_SAFE = 1 and it works. I use core 1 only to handle USB data so I don't use core 1 to write storage. Thanks!

mringwal, 2023-07-28:
@peterharperuk Can you follow the problem and the solution/workaround? Is there a general fix that could be applied?

wasdwasd0105, 2023-07-29:
I will leave the issue open