bluekitchen/btstack #501: why VMIN = 1，is this useless?

yangn0, 2023-07-12:
in the platform/posix/btstack_uart_posix.c 

line 309 , open() set the O_NONBLOCK. In non-blocking mode, VMIN/VTIME have no effect (POSIX specifications).

line 330 , why VMIN = 1?

Is this useless?

yangn0, 2023-07-12:
https://stackoverflow.com/questions/20154157/termios-vmin-vtime-and-blocking-non-blocking-read-operations

mringwal, 2023-07-12:
Hi. To be honest, I don't remember. Maybe I've used it without non-blocking first. Then, there might have been a reason (some USB-UART adapter, some USB-CDC device..) that didn't work with it and setting non-blocking made it work. Feel free to browse the version history.

Why did you stumble upon our uart driver?

mringwal, 2023-07-14:
In any case, it's not a bug :)

closed 2023-07-14 by mringwal

yangn0, 2023-07-20:
Sorry for replying so late.
thanks for you reply :)

I am porting btstack to a POSIX system. I have a problem about POSIX compliant.(https://devel.rtems.org/ticket/4926)
I want to make it clear once and for all.

mringwal, 2023-07-20:
The idea is that the run loop blocks with select() or similar until the fd becomes readable. Then, it calls read with a large buffer and expects that the read call will return immediately to not block. 

For write, we don't want it to block either. Given that there's hardware flow control which might prevent the write call from immediate success, the non-blocking option must stay. If we have non-blocking, then the VMIN setting is probably not needed.