bluekitchen/btstack #502: There is a wrong parameter type in gap_le_set_phy interface.

moyaoyi, 2023-07-13:
**Describe the bug**
Dear All:

There is an interface in **hci.c** as below:

uint8_t gap_le_set_phy(hci_con_handle_t con_handle, uint8_t all_phys, uint8_t tx_phys, uint8_t rx_phys, **uint8_t phy_options**);

**uint8_t phy_options** this parameter need modify to **uint16_t** type

Please refer to Core_v5.3.pdf - [BLUETOOTH CORE SPECIFICATION Version 5.3 | **Vol 4, Part E** page 2443]-[**7.8.49 **LE Set PHY**** command].

**phy_options** size is **2 octets**

**Best Regards!**

mringwal, 2023-07-14:
Thanks for reporting. The phy_options in LE Set PHY is 2 octets. In  Core v5.4, only the first two bits are used. I've updated the gap_le_set_phy to accept a uitn16_t for now on develop branch.

Why did you stumble upon this? :)

closed 2023-07-14 by mringwal

moyaoyi, 2023-07-21:
I think controller need an uint16_t type phy_option parameter for hci_le_set_phy comand.
If an uint8_t type phy_option might make controller enter exception because memory out of bounds access.

Best Regards!

mringwal, 2023-07-21:
That's correct. The command format wasn't correct either. it's fixed on master & develop now.

commit 9d8be771 (2023-07-14): gap: use uint16_t for phy_options in gap_le_set_phy, fixes #502