bluekitchen/btstack #503: there is a wrong iso stream state change in hci.c

moyaoyi, 2023-07-13:
**Describe the bug**

Dear All:

There is a wrong iso stream state changed in hci.c. It will cause setup ISO output failed after accept cis connection request.

<**master** branch>- <**hci.c**>-<function:  static void **handle_command_complete_event**(uint8_t * packet, uint16_t size)>
-<line 3022>-<**HCI_ISO_STREAM_STATE_W2_SETUP_ISO_INPUT** need modify to **HCI_ISO_STREAM_STATE_W2_SETUP_ISO_OUTPUT**>
some code as below:
````` 
                       switch (iso_stream->state){
                            case HCI_ISO_STREAM_STATE_W4_ISO_SETUP_INPUT:
                                if (status != ERROR_CODE_SUCCESS){
                                    emit_cis_created = true;
                                    break;
                                }
                                if (iso_stream->max_sdu_c_to_p > 0){
                                    iso_stream->state = HCI_ISO_STREAM_STATE_W2_SETUP_ISO_INPUT; ///< need modify to HCI_ISO_STREAM_STATE_W2_SETUP_ISO_OUTPUT
                                } else {
                                    emit_cis_created = true;
                                }
                                break;
`````

**Environment:  (please complete the following information):**
 - Current BTstack branch: [**master**]

Best Regards!

mringwal, 2023-07-14:
Thanks for reporting this. Indeed, this should be HCI_ISO_STREAM_STATE_W2_SETUP_ISO_OUTPUT.

Did you try to setup a bi-direction audio stream?

closed 2023-07-14 by mringwal

moyaoyi, 2023-07-21:
Did you try to setup a bi-direction audio stream?
>>
No, I did not. so sorry.

mringwal, 2023-07-21:
No problem. I was just curious how you found this and it looked like the code would work for either input or output, but not for both.

commit 1418332d (2023-07-14): hci: fix setup of bi-direction iso stream, close #503