bluekitchen/btstack #505: Multiple pairing fail until power off/on the host

amorniroli, 2023-07-17:
**Describe the bug**

I'm trying to "stress test" the BLE pairing feature.

The first pairing succeed; the next attempts will always fail due to 0x0B reason (SMP_SC_DHKEY_CHECK_FAILED). To succeed again, I have to power off / power on the board running the host (btstack).

**To Reproduce**

This is the loop I'm doing:

1. start scan;
2. find the device I want to connect to (in the wireshark trace, the device address is always 02:80:E1:00:42:28);
3. connect and start device info service (just for debug purpose);
4. when service query done, if key is already stored, delete it, disconnect, and go to step 1. Otherwise, go to step 5.
5. start pairing and when completed, go to 1.

If useful, I could share a 'standalone' example doing this behavior (just need a bit to clean the code and arrange it)

**Expected behavior**

I would expect that pairing would always succeed without the need to "power off / power on" the board (acting as a "clean" start for the btstack).

My guess is that some flag in sm.c remains in a wrong state between multiple pairings.

**HCI Packet Logs**
[multiple_pair.txt](https://github.com/bluekitchen/btstack/files/12068138/multiple_pair.txt) (rename to .pklg extension).

From N° 1 to N° 405 -> first pairing (succeed);
From N° 406 to N° 1475 -> multiple failing pairing;

At this point, I power off / on the board

From N° 1476 to end -> first pairing (succeed)

**Environment:  (please complete the following information):**
 - btstack v1.5.6.2-extra tag (https://github.com/bluekitchen/btstack/releases/tag/v1.5.6.2-extra)
 - Bluetooth Controller: ST BlueNRG2
 - Remote device: Another ST BlueNRG2 running as peripheral (I never power down this board).
 - [btstack_config.txt](https://github.com/bluekitchen/btstack/files/12068172/btstack_config.txt) (ENABLE_MICRO_ECC_FOR_LE_SECURE_CONNECTIONS)

mringwal, 2023-07-17:
Thanks for sharing this. The pairing failed on the second pairing is unexpected. 
What's running on the peripheral? (The peripheral sends the pairing failed, so it could also be an error on the remote side).

Anyway, if you have a standalone example, this would be very helpful. I guess I could run e.g. gatt_counter as the peripheral to test with.

amorniroli, 2023-07-18:
> Thanks for sharing this. The pairing failed on the second pairing is unexpected. What's running on the peripheral? (The peripheral sends the pairing failed, so it could also be an error on the [...]

It seems to be something related to peripheral. Using iOS as peripheral it's working. I close the issue and I'll try to investigate deeper by my side. 

Thank you.

Alessandro

closed 2023-07-18 by amorniroli