bluekitchen/btstack #506: it can't do connect audio stream successfully if undef HAVE_BTSTACK_STDIN' on hfp_ag_demo

alisagao123, 2023-07-18:
**Describe the bug**

it can't do connect audio stream successfully if undefine the HAVE_BTSTACK_STDIN on hfp ag side.

**To Reproduce**

Steps to reproduce the behavior:
1. build the hfp_ag_demo after undef the 'HAVE_BTSTACK_STDIN' in hfp_ag_demo.c
#undef HAVE_BTSTACK_STDIN
3. initilize the bluetooth on the hfp hands free unit,
4. run hfp_ag_demo on some Linux PC as below:
#sudo ./hfp_ag_demo 9C:50:D1:45:70:CA
5. on hfp hands free unit, do connection with this Linux PC which has USB bluetooth dongle:
root@imx8ulpevk:~# bluetoothctl
Agent registered
[CHG] Controller 9C:50:D1:45:70:CA Pairable: yes
[CHG] Device 00:1A:7D:DA:71:39 Connected: yes
[HFP AG Demo 00:1A:7D:DA:71:39]#
6. on the hfp ag side(this Linux PC),  it should do HFP service connection and audio stream connection automatically.
LOG as below:
alisa@Opt-5080:~/hfp_auto/btstack/port/libusb$ sudo ./hfp_ag_demo 9C:50:D1:45:70:CA
[sudo] password for alisa:
Packet Log: /tmp/hci_dump.pklg
connecting to 9C:50:D1:45:70:CA
SCO Demo: Sending sine wave, audio output via btstack_audio.
SDP service record size: 94
USB device 0x0a12/0x0001, path: 0d
libusb: error [udev_hotplug_event] ignoring udev action change
libusb: error [udev_hotplug_event] ignoring udev action change
Local version information:
- HCI Version    0x0006
- HCI Revision   0x22bb
- LMP Version    0x0006
- LMP Subversion 0x22bb
- Manufacturer   0x000a
TLV path: /tmp/btstack_00-1A-7D-DA-71-39.tlv
BTstack up and running on 00:1A:7D:DA:71:39.
Supported codecs: CVSD
Establish HFP AG service level connection to 9C:50:D1:45:70:CA...
==Alisa Debug==ag establish service level status=: (null)
Service level connection established to 9C:50:D1:45:70:CA.
Supported codecs: CVSD
Establish Audio connection 9C:50:D1:45:70:CA... ==> it stopped here and no any stream tranfer 
Alisa debug===audio establishing ===
Alisa debug===hfp_connection->negotiated_codec =0==
Alisa debug===hfp_connection->remote_codecs[0] =1==
Alisa debug===audio connection HFP_CODECS_IDLE =0==
hfp_ag_establish_audio_connection, status 0x00  

7. if no  undefine the HAVE_BTSTACK_STDIN, and the user anter 'a' and 'b', the SCO stream will tranfer successfully as below:
alisa@Opt-5080:~/hfp_auto/btstack/port/libusb$ sudo ./hfp_ag_demo_sdtin 9C:50:D1:45:70:CA
Packet Log: /tmp/hci_dump.pklg
connecting to 9C:50:D1:45:70:CA
SCO Demo: Sending sine wave, audio output via btstack_audio.
SDP service record size: 94
USB device 0x0a12/0x0001, path: 0d
libusb: error [udev_hotplug_event] ignoring udev action change
libusb: error [udev_hotplug_event] ignoring udev action change
Local version information:
- HCI Version    0x0006
- HCI Revision   0x22bb
- LMP Version    0x0006
- LMP Subversion 0x22bb
- Manufacturer   0x000a
TLV path: /tmp/btstack_00-1A-7D-DA-71-39.tlv
BTstack up and running on 00:1A:7D:DA:71:39.
Supported codecs: CVSD
Establish HFP service level connection to 9C:50:D1:45:70:CA...
Connection failed, status 0x04
Establish HFP service level connection to 9C:50:D1:45:70:CA...
Service level connection established to 9C:50:D1:45:70:CA.
Supported codecs: CVSD
Speaker volume: gain 7
Microphone volume: gain 7
Establish Audio connection 9C:50:D1:45:70:CA...
Audio connection established with SCO handle 0x0040.
Using CVSD codec.
SCO Demo: Init CVSD
SCO: sent 100, received 80
SCO: sent 200, received 180
SCO: sent 300, received 280
SCO: sent 400, received 380
SCO: sent 500, received 480
SCO: sent 600, received 580
SCO: sent 700, received 680
SCO: sent 800, received 780
CTRL-C - SIGINT received, shutting down..
Audio connection released
SCO demo close
SCO demo statistics: Used CVSD with PLC, number of proccesed frames:
 - 782 good frames,
 - 14 bad frames.
Service level connection released.
alisa@Opt-5080:~/hfp_auto/btstack/port/libusb$

**Expected behavior**
on the hfp ag side, it should do HFP service connection and audio stream connection automatically.

**HCI Packet Logs**
For all Bluetooth issues, please provide a full packet log that includes initial pairing. You need to zip the packet log as GitHub does not know how to handle .pklg files. 
[image]

alisagao123, 2023-07-18:
/

mringwal, 2023-07-20:
Thanks for reporting. Calling establish audio connection from the service level complete event in the HFP AG demo didn't work as the event was emitted before the last response to HF was sent. 

Please verify that it works with the latest version of the develop branch and let us know.

alisagao123, 2023-07-20:
@mringwal, it works if I apply your patch as below:
move the 'hfp_ag_slc_established(hfp_connection);' to follow the 'hfp_ag_send_retrieve_initital_supported_generic_status_indicators_cmd'
thanks~~
BTW, could you tell me how to open log_info function? I didn't find how to let it print the log.

mringwal, 2023-07-20:
The log_info as well the HCI trace is written to '/tmp/hci_dump.pklg' by default on POSIX systems. This is also shown in the console output.. ;) if you really want to print everything to the console, you can usean instance of hci_dump_posix_stdout instead of hci_dump_posix_fs in the call to hci_dump_init(...)

closed 2023-07-20 by mringwal

alisagao123, 2023-07-20:
@mringwal, another question:
how to create an array like this in sco_demo_util.c from any audio file? thanks~
// input signal: pre-computed int16 sine wave, 32000 Hz at 266 Hz
static const int16_t sine_int16[] = {
     0,   1715,   3425,   5126,   6813,   8481,  10126,  11743,  13328,  14876,
 16383,  17846,  19260,  20621,  21925,  23170,  24351,  25465,  26509,  27481,
 28377,  29196,  29934,  30591,  31163,  31650,  32051,  32364,  32587,  32722,
 32767,  32722,  32587,  32364,  32051,  31650,  31163,  30591,  29934,  29196,
 28377,  27481,  26509,  25465,  24351,  23170,  21925,  20621,  19260,  17846,
 16383,  14876,  13328,  11743,  10126,   8481,   6813,   5126,   3425,   1715,
     0,  -1715,  -3425,  -5126,  -6813,  -8481, -10126, -11743, -13328, -14876,
-16384, -17846, -19260, -20621, -21925, -23170, -24351, -25465, -26509, -27481,
-28377, -29196, -29934, -30591, -31163, -31650, -32051, -32364, -32587, -32722,
-32767, -32722, -32587, -32364, -32051, -31650, -31163, -30591, -29934, -29196,
-28377, -27481, -26509, -25465, -24351, -23170, -21925, -20621, -19260, -17846,
-16384, -14876, -13328, -11743, -10126,  -8481,  -6813,  -5126,  -3425,  -1715,
};

mringwal, 2023-07-20:
please post general questions the btstack mailing list - see readme.md.

the sine was created with tool/sine_table_generator.py
for a general audio file you would need to open it, decode it - .wav is simple, .mp3 is hard, or use some existing audio file library and then just print out the data. Audio files will become huge, that's why we use the mod player in our audio demos.