bluekitchen/btstack #508: how to send dtmf tone directly in hfp_ag_demo

alisagao123, 2023-07-21:
**Is your feature request related to a problem? Please describe.**

currently, it sends a sine wave which is defined in sco_demo_util.c sine_int16[]
can I do some modification to let it send dtmf tone directly in hfp_ag_demo

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mringwal, 2023-07-21:
If you have your device configured for SCO-over-HCI, you can send any sounds in either role. The sco_demo_util code also allows to send music (a mod song) next to microphone input or the test sine wave.

Why do you want to send DTMF sounds to the HF unit? There's usually a human with a headset on that side.

You can tell the AG to send DTMF sounds over the telephone network by calling hfp_hf_send_dtmf_code on the HFP HF side. This might be useful to navigate some call center navigation system.

alisagao123, 2023-07-21:
thanks for your quick response.
actually, I want to do bluetooth HFP call function automated testing.
My target test board works as HFP hands free unit, 
and I loopback the headset and micphone in hands free unit when hfp_ag_demo is running on one Linux PC,
and in hfg_ag_demo side, I will check if the output file 'sco_input.wav' has noise or no voice to verify if hands free unit works well.
it is easy for me to check if dtmf tone has voice or no voice, it has mature algorithms to do this. 
I don't want to change anything in hands free unit as I need to test it.

mringwal, 2023-07-21:
Still not sure what you want to do. Could you clarify your feature request? Feel free to discuss general ideas on the mailing list or get in touch with us for engineering support.

HFP calls work by sending a AT command from HF to AG - no sound there. In the HFP AG demo, you should receive an event for the 'initiate call to 12345678' and also events if the user presses buttons on the HF side during the call.

If you just need any sound, you could use our mod player, or play sine tones of different frequencies.

alisagao123, 2023-07-21:
[image]

Hi @mringwal , I uploaded an picture, I am not sure if you can get my idea. thanks.
and I just want to do some simple modification in source code.

alisagao123, 2023-07-21:
Is the voice I heared on hands free unit side from the array defined in sco_demo_util.c sine_int16[]? I thought it was.
can I just do some modification in this array , and then I can hear the dtmf tone on hands free unit side?

closed 2023-07-21 by mringwal

alisagao123, 2023-07-24:
when I modify the SCO_DEMO_MODE to mod player, there are many failures when building.
// SCO demo configuration
//#define SCO_DEMO_MODE               SCO_DEMO_MODE_SINE
#define SCO_DEMO_MODE               SCO_DEMO_MODE_MODPLAYER

alisagao123, 2023-07-24:
In source code, where I can set the audio card if I define the SCO_DEMO_MODE to microphone?
as I insert an external usb audio card into my Linux PC, I don't want to use the default one on this PC,
 #define SCO_DEMO_MODE    SCO_DEMO_MODE_MICROPHONE

mringwal, 2023-07-25:
The posix port uses the default configuration of PortAudio. The examples avoid any knowledge of the underlying system. 

Sample rate / num channels is fixed for Bluetooth audio. 

Feel free to patch sco_util.c and/or platform/posix/btstack_audio_portaudio.c as needed.

alisagao123, 2023-07-28:
thanks @mringwal ,
is there example for obex file transfer?

mringwal, 2023-07-29:
OBEX File Transfer - Object Push Profile - is available to commercial customers. 

If you use BTstack as some kind of end-of-line product test in manufacturing, this classifies as commercial use which requires a license. 

Please get in touch with us at contact@bluekichen-gmbh.com for details.