bluekitchen/btstack #512: Example le_credit_based_flow_control_mode_server contains uncompilable code

felixdoerre, 2023-07-26:
**Describe the bug**

Compiling `le_credit_based_flow_control_mode_server` with `TEST_STREAM_DATA` does not work. 

**To Reproduce**

Steps to reproduce the behavior:
1. Compile example 'le_credit_based_flow_control_mode_server' with `-DTEST_STREAM_DATA=1`

**Expected behavior**

The example compiles, demonstrating correct cbm-API usage.

**Observed behavior**
The example uses (presumably old) API functions which don't exist. The code hidden behind this preprocessor flag should either be updated or removed/clearly marked as non-functioning.
```
../../example/le_credit_based_flow_control_mode_server
../../example/le_credit_based_flow_control_mode_server.c: In function 'streamer':
../../example/le_credit_based_flow_control_mode_server.c:139:9: error: 'le_data_channel_connection' undeclared (first use in this function); did you mean 'le_cbm_connection'?
     if (le_data_channel_connection.cid == 0) return;
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
         le_cbm_connection
../../example/le_credit_based_flow_control_mode_server.c:139:9: note: each undeclared identifier is reported only once for each function it appears in
../../example/le_credit_based_flow_control_mode_server.c:147:5: error: implicit declaration of function 'l2cap_cbm_send_data'; did you mean 'l2cap_le_send_data'? [-Werror=implicit-function-declaration]
     l2cap_cbm_send_data(le_data_channel_connection.cid, (uint8_t *) le_data_channel_connection.test_data, le_data_channel_connection.test_data_len);
     ^~~~~~~~~~~~~~~~~~~
     l2cap_le_send_data
../../example/le_credit_based_flow_control_mode_server.c: In function 'packet_handler':
../../example/le_credit_based_flow_control_mode_server.c:250:62: error: 'le_data_channel_connection' undeclared (first use in this function); did you mean 'le_cbm_connection'?
                         l2cap_cbm_request_can_send_now_event(le_data_channel_connection.cid);
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                                              le_cbm_connection
../../example/le_credit_based_flow_control_mode_server.c:263:22: error: 'L2CAP_EVENT_CBM_CAN_SEND_NOW' undeclared (first use in this function); did you mean 'L2CAP_EVENT_CAN_SEND_NOW'?
                 case L2CAP_EVENT_CBM_CAN_SEND_NOW:
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                      L2CAP_EVENT_CAN_SEND_NOW
```

felixdoerre, 2023-07-26:
Also note that the function that `gcc` is recommending (`l2cap_le_send_data`) has a deprecation annotation https://github.com/bluekitchen/btstack/blob/de0ec9eb80b59ba1d83d142547fa900afb1f5f00/src/l2cap.h#L845 referencing `l2cap_send_data` which I cannot find.

mringwal, 2023-07-27:
Thanks for reporting. Looks like we've missed the code disabled by the missing TEST_STREAM_DATA when streamlining the API names. Please verify the fix on the develop branch.

felixdoerre, 2023-07-27:
Thx, the example compiles now. And I see you've adjusted 2 deprecation annotations, but not the one in line 845 that I referenced. The function now seems to be called `l2cap_send` instead of `l2cap_send_data`

mringwal, 2023-07-28:
Thanks. I've fixed this and another one on develop.

closed 2023-07-28 by mringwal