bluekitchen/btstack #516: btstack_tlv_flash_bank wrong alignment for read/write offsets

amorniroli, 2023-08-01:
**Describe the bug**

Read/write and migrate routines are not taking in consideration correct alignment of the target.

For example, STM32U575 MCU has flash alignment of 128bit (16-byte alignment). In some part of the btstack_tlv_flash_bank code offset increment is done considering 8-byte, which will cause write failure and/or wrong values when reading.

**To Reproduce**

Using #define BTSTACK_FLASH_ALIGNMENT_MAX 16 will cause failures on read/write routines.

**Expected behavior**

Routines should work on all "types" of flash alignments.

**HCI Packet Logs**
Not needed

**Environment:  (please complete the following information):**
File modifed starting from btstack-1.5.6.2-extra

**Additional context**
Attached (rename to .c) the modified implementation which is working (at least, it seems to work) on the STM32U575 (I may have missed something, but it could be a start point to check the routines).
[btstack_tlv_flash_bank.txt](https://github.com/bluekitchen/btstack/files/12226151/btstack_tlv_flash_bank.txt)

mringwal, 2023-08-01:
Thanks for reporting. I do vaguely remember that the alignment works only for 1,2,4,8 bytes, but don't see any documentation for this. I'll have a look at this and also your suggested fix when we are back from vacation next week.

mringwal, 2023-08-29:
Hi Alessandro. We're back and I've reworked the flash bank tlv implementation. Changes:
- delete tag for alignment 8 and larger: good idea to fully erase the complete entry.
- When using an explicit delete field, the entry header needs to be padded
- Without an explicit delete field, padding the entry header seems rather wasteful. I've optimized the code to let the value always start after the 8 bytes { tag, len } header even with larger alignments. This allows for up to 8 bytes of value stored in a 16-bytes block. Up to 24 bytes of data can be stored in 2x16-byte blocks.

Could you test/verify the code on develop on your MCU?
Our unit test with alignment still works, so there's a fair chance the new code will also work with alignment = 16.

amorniroli, 2023-08-30:
> Hi Alessandro. We're back and I've reworked the flash bank tlv implementation. Changes:
>
> * delete tag for alignment 8 and larger: good idea to fully erase the complete entry.
> * When using an [...]

Hi @mringwal ,

I've just pulled develop branch and tried, but there is still a bug in function btstack_tlv_flash_bank_read, line 134:

```
static void btstack_tlv_flash_bank_read(btstack_tlv_flash_bank_t * self, int bank, uint32_t offset, uint8_t * buffer, uint32_t size){

	// read main data
	uint32_t alignment = self->hal_flash_bank_impl->get_alignment(self->hal_flash_bank_context);
	uint32_t lower_bits = size & (alignment - 1);
	uint32_t size_aligned = size - lower_bits;
	if (size_aligned){
		self->hal_flash_bank_impl->read(self->hal_flash_bank_context, bank, offset, buffer, size_aligned);
		buffer += size_aligned;
		offset += size_aligned;
		size   -= size_aligned;
	}

	// read last part
	if (size == 0) return;
	uint8_t alignment_block[BTSTACK_FLASH_ALIGNMENT_MAX];
	self->hal_flash_bank_impl->read(self->hal_flash_bank_context, bank, offset, alignment_block, alignment);
//	uint32_t bytes_to_copy = btstack_min(alignment - lower_bits, size);
	memcpy(buffer, alignment_block, size);
}
```

When not aligned, at the end just copy the remaining size to read. bytes_to_copy is not needed. With this fix seems to work as expected: I'll do some more tests today, trying also migration.

Thank you!

mringwal, 2023-08-30:
Thanks for the feedback. You're right in both points. 

Btw. BTSTACK_FLASH_ALIGNMENT_MAX only needs to be >= the alignment from the lower layer, e.g. you could set it to 32 and have self->hal_flash_bank_impl->get_alignment(..) return 16 and it would align for 16 byte reads/writes.

mringwal, 2023-09-18:
Have you been able to check if the updated/current version works for you now? Thanks!

amorniroli, 2023-09-26:
> Have you been able to check if the updated/current version works for you now? Thanks!

Hi @mringwal,

just came back from vacancy. I'll try in these days and let you know.

mringwal, 2023-11-10:
I'll assume that it's working as expected. Feel free to re-open if needed. :)

closed 2023-11-10 by mringwal